prusa3d/PrusaSlicer #476: Make the shortcuts case insensitive

m-Th, 2017-08-15:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-win64
Build: Slic3r-1.36.2-prusa3d-win64-full-201707281218
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601]

### Behavior
Rotating objects +/-45 via keyboard shortcuts 'l' and 'r' doesn't work with 'L' and 'R' respectively. Please make them case insensitive.

Recommended fix: when you read the keystroke, "filter" it via a function which transforms it to lowercase. Applicable to all shortcuts, of course.

_Is this a new feature request?_
Nope. A small bug.

bubnikv, 2017-08-15:
This is intentional.

closed 2017-08-15 by bubnikv

m-Th, 2017-08-15:
<head explodes>