prusa3d/PrusaSlicer #477: Multi Material Single Mode - Beep before asking for Extruder

m-Th, 2017-08-15:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-win64
Build: Slic3r-1.36.2-prusa3d-win64-full-201707281218
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601]

### Behavior
The Start G-Code for Multi Material Single Mode starts like this:

M115 U3.0.12 ; tell printer latest fw version
; Start G-Code sequence START
T?
M109 S[first_layer_temperature]
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G28 W
...

However the time till "T?" is rather long because the heatbed needs to reach the appropriate temperature for the print. That's why it would be very appropriate to add a Beep before the "T?" question in order to inform the user that the printer awaits a response from him. Hence we should have:

M115 U3.0.12 ; tell printer latest fw version
; Start G-Code sequence START
**M300 S300 P800 ; beep**
T?
M109 S[first_layer_temperature]
G21 ; set units to millimeters
G90 ; use absolute coordinates
M83 ; use relative distances for extrusion
G28 W
...

_Is this a new feature request?_
yes

bubnikv, 2017-08-15:
@jindrichbenes Do we want the beep to be added to 
https://github.com/prusa3d/slic3r-settings ?

jindrichbenes, 2017-08-29:
Latest settings were changed to 

M115 U3.0.12 ; tell printer latest fw version
; Start G-Code sequence START
T?
M104 S[first_layer_temperature]
M140 S[first_layer_bed_temperature]
M109 S[first_layer_temperature]
M190 S[first_layer_bed_temperature]
G21 ; set units to millimeters

so printer will ask for extruder immediately after selecting gcode, then starts to warm up.

m-Th, 2017-08-29:
@jindrichbenes - ok, this is another solution which fits for me. ...perhaps a better one because if we have a printer farm it will be difficult to know which one beeps...

bubnikv, 2017-08-29:
thanks, closing this issue.

closed 2017-08-29 by bubnikv