prusa3d/PrusaSlicer #478: Z Offset offset

m-Th, 2017-08-15:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-stable-win64
Build: Slic3r-1.36.2-prusa3d-stable-win64-full-201707281104
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601]

### Behavior
The Z offset label is not on the same line with the corresponding Edit Box.

See here:

[image]

_Is this a new feature request?_
Nope. A small GUI glitch.

bubnikv, 2018-02-12:
@YuSanka is currently working on this piece of code. @YuSanka please try to fix it.

YuSanka, 2018-02-15:
Fixed with 752d089

closed 2018-02-15 by YuSanka