prusa3d/PrusaSlicer #480: New feature request - Additional "Filament Settings" for retraction

jslick007, 2017-08-15:
### Version
1.36.2

### Operating system type + version
Windows 10

### Behavior
Add ability to override retraction settings set in "Printer Settings" for different filaments. Standard "Printer Settings" work for most filaments, but for PET-G, would be nice to override just for that "Filament Profile" and not the entire printer.

ex1. Printer settings set for .8 mm retraction, Filament settings set for 1.4mm retraction, print comes out at 1.4mm retraction.

ex2. Printer settings set for .8 mm retraction, no filament settings, print comes out at .8 mm retraction.

m-Th, 2017-08-21:
Yes! Fully agree! Absolutely! Upvoted! Support! Just beat me on posting this. 

Also, I think that it is obvious: Retraction is dependent of Filament's nature, and not on Extruder number (!) like it is now.

I can provide different Retraction values for different materials.

From implementation, UX/GUI standing point I think that the easiest and most natural implementation would be to carry the entire (sub)class and corresponding GUI which holds the Retraction engine. Of course, a "Retraction override enabled" checkbox should be added. If this is ON then the values from Filament will take precedence. If it is OFF the global, extruder/printer will be in effect:

[image]

bubnikv, 2017-08-28:
> Also, I think that it is obvious: Retraction is dependent of Filament's
nature, and not on Extruder number (!) like it is now.

It is dependent on both the extruder train and the filament.

m-Th, 2017-08-28:
> It is dependent on both the extruder train and the filament.

...But since we have the same extruder train for all 4 extruders, this factor disappears, isn't it?

bubnikv, 2017-08-28:
You may want to use a single set of filament profiles with multiple different printers.

jslick007, 2017-08-28:
> You may want to use a single set of filament profiles with multiple different printers.

I do this now, but it would be easier, more intuitive, and make more sense to store the value with with the filament itself.

rongith, 2017-09-26:
Alexrj implemented an override feature: https://github.com/alexrj/Slic3r/issues/3643
That is nice to me.

hoegge, 2018-01-24:
So - any news on this. Nice that it is implemented in Slic3r but that does not really help in Slic3r PE. Any prospect on getting a solution for the i3 MM?

bubnikv, 2018-01-25:
We have a new hire @YuSanka currently working on re-write of this part of code into C++. We may ask her to implement this "filament retract override" feature once she stabilizes her current work.

parecek, 2018-02-12:
In original Slic3r is this feature implemented as overrides.... https://github.com/alexrj/Slic3r/issues/3643

Lyr3x, 2018-03-18:
Any news on this? Would really appreciate this feature 👍

bubnikv, 2018-03-18:
We are looking into something like this. We are currently collecting requirements, as of which parameters to allow overriding inside the filament profile.

ayberkozgur, 2018-09-04:
Any update on this? PETG filaments needing extra retraction is a crucial use case for this improvement.

Flijfi, 2018-10-22:
I'd like to upvote for this too. Please consider it among the priorities. Thanks

wavexx, 2018-11-21:
As far as requirement goes, I also have to tweak the bridge flow ratio depending on the material.

jonnieZG, 2018-12-06:
> From implementation, UX/GUI standing point I think that the easiest and most natural implementation would be to carry the entire (sub)class and corresponding GUI which holds the Retraction engine. [...]

There is no need for the checkbox. We should just set a value to override it, or leave it empty to inherit from printer settings.

bloodygoodbloke, 2019-01-14:
UPvote for me too.

oddlots, 2019-05-09:
+1 please

chschmit, 2019-05-29:
+10 please
I ended up making multiple Printer settings with Filament type in the title so that I make sure to align the Filament and Printer settings to the get correct retraction or Z offset for PETG. This is a hack in my opinion and need to be handled as a defined setting or override setting.

ayberkozgur, 2019-06-03:
Any news on this?

bubnikv, 2019-06-03:
It has a relatively high priority.

bubnikv, 2019-08-01:
retract overrides at the filament profile are implemented by the upcoming PrusaSlicer 2.1.0-alpha1.

closed 2019-08-01 by bubnikv

sticilface, 2020-04-02:
This issue was closed: Feature Request: Adding custom settings to filaments. #2538 as a duplicate of this feature here, which is now closed as 2.1 contains filament overrides.  However, the crucial feature discussed in BOTH these requests has NOT been implemented.  namely:  Filament extrusion speeds.  I need this for TPU.. and as discussed I have to make an entire new printer profile for TPU use.  I am also using TPU and PETG in the same print using the MMU and have to stick to the super slow TPU speeds when it should speed up for PETG. 

Can this be added in addition to retraction overrides please?

jonnieZG, 2020-04-03:
> This issue was closed: Feature Request: Adding custom settings to filaments. #2538 as a duplicate of this feature here, which is now closed as 2.1 contains filament overrides. However, the crucial [...]

I second that. Since this issue is already closed, I think you should open a new one just for this retraction speed override feature.