prusa3d/PrusaSlicer #485: When items are dragged near the center, they vanish.

jbum, 2017-08-20:
### Version
1.36.2-prusa3d

### Operating system type + version
macOS 10.12.6

### Behavior
When items are dragged close to the center of the print bed, they vanish.  Once vanished, no amount of dragging gets them back.

Sometimes shortly after this the print bed disappears as well.

This happens with most of the STLs I've used.  Happens a lot when I'm duplicating and arranging multiple items.  Very hard to get 5 or 6 duplicates arranged without one or more of them disappearing.  I'm wondering if I'm getting a divide-by-zero when items are dragged through the center...

m-Th, 2017-08-21:
Cannot reproduce on Win7. Perhaps it is better to upload (if you can) some STLs?

numanair, 2017-08-21:
@jbum do you have a bed with the center as zero? I've experienced this issue (I don't have a center as zero bed though). I suspect they are getting moved out of sight as sometimes I'll see an object jump to beside the bed area.

jbum, 2017-08-21:
m-Th: Can't upload stl to github, but here's a link to one I posted on my server: 

https://krazydad.com/icosahedron_tile_v2.stl

Like I said, this happens with pretty much every stl I've used.

numanair: unsure if the center is zero, but the axis display is in the corner.  When the vanish happens, I've tried zooming way out to see if they went somewhere distant, but they don't show up.  Very often, the bed display disappears soon after, and I have to restart.

jbum, 2017-08-21:
Also note it happens more frequently when I duplicate an item a few times and move the copies around.

m-Th, 2017-08-22:
@jbum : Sorry, my firewall sez that your site is blocked. :) (Games). You can zip it and post it here. :)

jbum, 2017-08-22:
[icoso_tile_v2.stl.zip](https://github.com/prusa3d/Slic3r/files/1241220/icoso_tile_v2.stl.zip)
There ya go

m-Th, 2017-08-22:
Works ok on win7. Sorry. :)

In Slic3r go to Printer Settings tab \ General and click on Bed Shape... what values for Shape, Size and Origin do you have?

jbum, 2017-08-22:
It's been preset to a custom bed using the Prusa i3 initialization - the actual measurements aren't shown, but it looks like this:

[image]

jbum, 2017-08-22:
Hey, I just noticed I was using the 'disabled' settings 'Slic3r settings 175' from that archive, instead of the more recent Slic3r settings MK2.  I will switch to that, and see if it fixes it.

bubnikv, 2017-09-12:
I confirm that this is a bug specific to the custom bed shape.

bubnikv, 2017-11-30:
I have just tested with the 1.38.3 alpha (not published yet) and it seems the problem fixed itself (or likely some other bugfix fixed this issue). Would you please re-test with the upcoming alpha release?

bubnikv, 2018-02-12:
No response, closing. It seem to be fixed.

closed 2018-02-12 by bubnikv