prusa3d/PrusaSlicer #486: Printing in Manual Order

m-Th, 2017-08-21:
### Version
Slic3r Prusa Edition
Version: 1.36.2-prusa3d-stable-win64
Build: Slic3r-1.36.2-prusa3d-stable-win64-full-201707281104
Operating System: MSWin32
System Architecture: MSWin32-x64-multi-thread
Windows Version: Microsoft Windows [Version 6.1.7601] 

### Behavior
We need a way to select manually the order in which different objects will be printed on plater. This is needed because at least of two important factors:

* We want to use sequential printing. For this to work we must choose manually the order of objects to be printed because otherwise there is danger to crush with the nozzle / hotend the objects already printed.

* The algorithm for inter-object path planning is very rudimentary (at best) - see https://github.com/prusa3d/Slic3r/issues/409 .

_Is this a new feature request?_
yes

bubnikv, 2019-08-01:
similar to https://github.com/prusa3d/PrusaSlicer/issues/810

wavexx, 2019-10-26:
I would argue ordering objects by Y then X whenever sequential printing is enabled would be much simpler as a start.

bubnikv, 2020-02-04:
The upcoming PrusaSlicer 2.2.0-alpha4 will print sequentially in the order displayed at the object list. Also the object list will support drag & drop to reorder.
Closing.

closed 2020-02-04 by bubnikv

Jaxx005, 2022-01-02:
Objects (not instances - see #6474) in the list can now be re-ordered by dragging, but the position/order when Auto-Arranged still seems incomplete/unclear?  

Example (Prusa Mk3S, Win10):  3 small objects inserted...
Arrange = Collision and/or illogical/unclear positioning on the bed.  
Reordered Objects in List + Arrange = as above -or- sometime positions unaltered.

The 'positioning' problem is same as #6474 it seems?