prusa3d/PrusaSlicer #488: [Request] Alternate extruder per layer

sarf2k4, 2017-08-23:
Is there any way that I can do alternate extruder colour during printing? As in I have 4 extruder, I thought that alternating layer colour would be cool since the one alternating is the prime pillar.

This also might able to produce some desired colour effect. For example:

layer 1 is e1
layer 2 is e2
layer 3 is e3
layer 4 is e4
layer 5 is e3 
layer 6 is e2 
layer 7 is e1

or 

layer 1 is e1
layer 2 is e2
layer 3 is e3
layer 4 is e4
layer 5 is e1 
layer 6 is e2 
layer 7 is e3
layer 8 is e4

either way, they can produce a cool looking gradient colour. I did this without prime pillar on simplify 3d, where there are 2 objects in a plate, it will print main body with extruder 1, while the other part was on extruder 2. by not using the prime pillar it will extruder the previous colour on to the designated part it prints.

sarf2k4, 2017-09-20:
To make things bit simple, this property setting will be placed on model properties where each model can have their own alternate layers.

There will be a text box for example
```
1 //this will assign the model will be fully printed with extruder 1

1,2,3,4 //this will tell the printer to alternate between layers just like the previous post's example
```

rtyr, 2021-01-20:
Already implemented. Closing.

[image]

closed 2021-01-20 by rtyr

JohnOCFII, 2021-01-21:
I was not aware of this feature.  I was just looking for that `Set extruder sequence` feature in PrusaSlicer 2.3, and even with searching for the term sequence, I can't find it.    I'm sure it is right in front of me.

rtyr, 2021-01-21:
In preview mode:
[image]

JohnOCFII, 2021-01-21:
Thanks @rtyr !