prusa3d/PrusaSlicer #489: Feature Request: Allow x and y portion of layer change to precede the z and retraction portion

akromatikus, 2017-08-23:
### Version
Slicer Prusa Edition V1.36.2

### Operating system type + version
OS X El Capitan V10.11.4

### Behavior
Hi all, 

I've been having problems with the z seam warping and blobbing in slicer on this calibration plate I made from hatchbox white pla: 
[image]
[image]

the wall is 1mm wide and 5mm tall, and with a .5mm extrusion width is made with 2 passes with layer start and end in the same spot. I have tried changing every setting in slicer with no substantial improvement from any, so I tried printing the plate with cura and the problem virtually vanished:
[image]
[image]

So I investigated by matching cura and slicer settings as closely as possible, and the only noticeable difference between the programs, is that at the end of a layer, cura will make the y move before making the z move and retraction, while slicer does it backwards. 

So feature request to allow the option of the x and y move portion of layer change to precede the z move and retraction portion.

#### STL/Config (.ZIP) where problem occurs
[calibrationPlate.stl.zip](https://github.com/prusa3d/Slic3r/files/1244080/calibrationPlate.stl.zip)
[calibationPlateSlicerConfig.ini.zip](https://github.com/prusa3d/Slic3r/files/1247288/calibationPlateSlicerConfig.ini.zip)

bubnikv, 2017-08-28:
I know Simplify3D does it the same way as you described. Slic3r jumps up
first and then moves in the XY plane to avoid kicking into the possibly
warped-up print.

Would allowing a wipe help?

akromatikus, 2017-08-30:
I actually had wipe on previously but it worsened the problem. I tried in again with the newer settings just in case and it did improve a good amount this time, possibly because my retraction speed is much slower now:
[image]
[image]

However, it created new problems. It might be hard to make out in the pic but the z dimension at the seam expanded and there is a blob at the top there. Mostly though, the issue is that that 5mm tall wall becomes warped/bent along its length! Without wipe enabled the wall stays nice and straight. I couldn't get a decent pic but it is obvious with the eye. 

So all in all the Simplify3d/cura method seems like a more elegant solution with less compromise. It seems the faster the extruder leaves the most recent extrusion and starts the next layer, the better. 

For the snagging problem, maybe have an option that makes x/y precede the z/retraction only if the beginning of the next layer is within a a specified distance from the ending of the previous layer?

bubnikv, 2017-10-03:
> So all in all the Simplify3d/cura method seems like a more elegant solution with less compromise. It seems the faster the extruder leaves the most recent extrusion and starts the next layer, the better.

I think it is machine specific. What is your printer?

It seems to me that maybe your machine has insufficient print cooling. For PLA you should not have such issues with the wipe.

If you have a bowden setup, you may also experiment with the 'Retract amount before wipe' value to do a fast partial retract before wipe, which also shortens the wipe move.

akromatikus, 2017-11-12:
I have the Prusa i3 MK2S.

The fans are ok I think. I have tried using various cooling settings from 100% and down and with the lowest and highest extruder temp I could manage but it has not made a significant difference. I do get a lot of oozing after travel moves and the extruder does not always break away cleanly from the extrudered line and I get strings. So far, nothing seems to help much so It could be the plastic.

The only hardware issue I'm aware of is that the rods are slightly bent. I had a lot of trouble getting the y axis to move smoothly after inserting the rods and ended up buying drylin bearings thinking it was the stock bearings. They helped but It is still quite resistant to movement and it turns out that when the plate moves all the way to the back a gap forms between a bearing and the rod. Now the software has noticed this. I can see and hear the z axis adjusting when the plate moves around that area so I assumed it was not an issue. The plate does wobble a bit when I press down on a corner though and It made me wonder if the extruder moves the plate slightly after a z move and this is worsening the z seam (I doubt it though). But like the pictures show it is mostly better now but I thought I'd mention the rods just in case it is indeed pertinent or a concern in general (should I get the rods replaced?).

I have had other problems however that may be related or suggest some other problem:

Tight corners over extrude but I assume that is pretty common?

The 2 main issues are:

The adjust X:Y setting tapers the wall and the change is uneven. In other words, what should be a ~1mm thick wall end up being a wall that's ~1.2mm at one end and smoothly tapers down to ~.9mm on the other end. This happens in Cura as well.

The other issue is that the fill density does nothing. 100% to say 5% makes little difference in the fill. At first I tried small changes in density and saw small changes, but I think these were just inconsistencies in printing and I didn't notice until later (on that note, I notice the first print of the day would under extrude compared to the rest. Is this Normal?)

So what can I do? The thread you linked seems similar but with these other issues I'm facing should I reinstall the firmware?

closed 2026-09-29 by prusaslicer-bot[bot]