prusa3d/PrusaSlicer #490: bad slicing on thin parts

aligator, 2017-08-24:
### Version
Slic3r 1.36.2 - Prusa Edition

### Operating system type + version
Linux Mint 18.2 Sonya x64

### Behavior
The attached model is a part of the wing from https://www.thingiverse.com/thing:2236583
if I print it vertically, the slicer cannot handle the thin part on the sides of the wing

[image]

Cura does this much better:
[image]

#### STL/Config (.ZIP) where problem occurs
[DG-101.zip](https://github.com/prusa3d/Slic3r/files/1248247/DG-101.zip)

Hugoz12, 2017-08-26:
Try disabling "Detect thin walls" in the Print settings: Layers and perimeters.

aligator, 2017-08-28:
Thank you. It works.

closed 2017-08-28 by aligator