prusa3d/PrusaSlicer #491: [FR] Option to extend brim internally only...

fiveangle, 2017-08-26:
For printing "pipe" like objects where a perimeter is printed on the bed surface and extends in the Z direction, a brim is often needed.

It would be extremely handy to have a toggle in the _Skirt and Brim_ section of _Print Settings_ to allow the brim feature to extend from the inside of the perimeter rather than the outside of the model because the majority of these designs have the internal perimter of the "pipe" not showing in the final piece.  This makes cleanup of brim easy instead of tedious because we can just rip the brim off indiscriminately since any visual artifcacts are hidden once the part is put into service.

I know you are very busy with high-priorty items but I have been wishing I had this feature a long time and just had yet another design fail by being knocked off the bed by the print head in the final milimeters due to me hesitant to use a brim because I did not want to have to clean it off the final piece.

This feature would be extremely nice to have.  Thanks for considering.

-=dave

bubnikv, 2017-08-28:
This issue is similar to https://github.com/prusa3d/Slic3r/issues/61
I know the mainline Slic3r has an "internal brim" settings, but the
implementation is questionable. I would have to do it differently, in a
similar fashion to the perimeter generator. I am planning to do this
feature together with the rework of the perimeter generator.

stepikovo, 2022-04-13:
Implemented in PrusaSlicer 2.4.0. Closing.

closed 2022-04-13 by stepikovo