prusa3d/PrusaSlicer #492: Soluble Interface Printing in Wrong Material (PLA)

KeelinJ, 2017-08-26:
Slic3r Prusa Edition 1.36.2
macOS Sierra 10.12.6

I'm having an issue slicing models with the 0.20mm NORMAL SOLUBLE INTERFACE preset (no modifications). It appears correctly in the Slic3r preview, with 4 layers of soluble interface (shown in red) on top of the support structure (shown in orange). However, when it prints, it prints some of the top interface layers in PLA, thus fusing the supports to the model. When it does this, it will successfully load the soluble material, then immediately unload it and load PLA before leaving the purge tower area. I've had this happen on two different models so far.

**Example 1:**
[image]
You can see the support structure with the soluble interface layers in this Slic3r preview screenshot.

[image]
However, those interface layers all printed in PLA, as seen in this photo.

**Example 2:**
[image]
This screenshot also appears to show the soluble interface layers correctly.

[image]
This time the interface layers started printing in the correct material, though they eventually printed in PLA. I cancelled the print as soon as I saw the issue.

I have attached my STLs and gcodes. Please let me know if there's anything else you need from me.
[soluble_interface_bug.zip](https://github.com/prusa3d/Slic3r/files/1253983/soluble_interface_bug.zip)

-Keelin Jacobsen

P.S. Thank you for all your hard work on this project. It is much appreciated.

m391118908, 2017-08-29:
Have the same issue, you can Preview the vault if you load the generated gcode in S3D

KeelinJ, 2017-09-21:
Still present in version 1.37.1

m391118908, 2017-09-22:
yes, still present in 1.37.1

landis90, 2017-09-28:
Have the same issue with only soluble interface feature.  
[image]
[image]

MStylesMS, 2017-10-02:
I also have this issue.

bubnikv, 2017-10-03:
Thanks for the report. I am looking into this issue, I am able to reproduce it.

bubnikv, 2017-10-03:
I hope I have fixed the bug with 
https://github.com/prusa3d/Slic3r/commit/eb0117b1c12d6bda5efc5856d7241d4b074f5876

I think I will provide a test build here.

bubnikv, 2017-10-03:
Here are the alpha builds. Please note that they are the development builds, far from being  tested thoroughly.

Windows 32bit/64bit
[Slic3r-1.37.1.10-prusa3d-win32-full-geb0117b-201710031130.zip](https://github.com/prusa3d/Slic3r/files/1352140/Slic3r-1.37.1.10-prusa3d-win32-full-geb0117b-201710031130.zip)
[Slic3r-1.37.1.10-prusa3d-win64-full-geb0117b-201710031130.zip](https://github.com/prusa3d/Slic3r/files/1352141/Slic3r-1.37.1.10-prusa3d-win64-full-geb0117b-201710031130.zip)

OSX
[Uploading Slic3r-1.37.1.10-prusa3d-full-geb0117b-201710031126.dmg.zip…]()

bubnikv, 2017-10-03:
[Slic3r-1.37.1.10-prusa3d-full-geb0117b-201710031126.dmg.zip](https://github.com/prusa3d/Slic3r/files/1352159/Slic3r-1.37.1.10-prusa3d-full-geb0117b-201710031126.dmg.zip)

KeelinJ, 2017-10-03:
The OSX version crashes every time I try to open it. I will try the PC version soon.

Thanks for taking a look at this, @bubnikv!

KeelinJ, 2017-10-04:
I just re-sliced my model in Windows and the gcode preview looks much better in S3D. I will run a print tomorrow to see for sure.

KeelinJ, 2017-10-04:
Now I'm running into some different issues. The temperature and speed no longer change for the soluble material and the fan doesn't turn off, which makes the material come out bubbly.

Also, some non-interface support layers are printing in soluble material for no apparent reason.

MStylesMS, 2017-10-04:
I am just seeing this.  bubnikv, should I go ahead and test the OS X version or wait for you to address the issue KeenlinJ brought up about the temperature and speed?

MStylesMS, 2017-10-10:
@bubnikv any progress on @KeelinJ post about the new issues?  I'm ready to test whenever you have a new build that addresses those.

24c, 2017-10-13:
OSX version 1.37.1.10 opens fine for me using macOS High Sierra. :?

bubnikv, 2017-10-21:
@KeelinJ 
> Now I'm running into some different issues. The temperature and speed no longer change for the soluble material and the fan doesn't turn off, which makes the material come out bubbly.

Would you please be more specific?

I realized that the fan may not be correctly controlled over the wipe tower. Other than that it shall work.

> Also, some non-interface support layers are printing in soluble material for no apparent reason.

Do you have the support material parameters set as on the following screenshot? The "soluble material" is important, as it basically tells the Slic3r: "This material is expensive, use it sparingly". If the checkbox is off, it means "This materal is as expensive as the others, try to minimize tool switches for faster prints".

[image]

I am going to release a patch build soon, so a quick response would be appreciated.

bubnikv, 2017-10-21:
@KeelinJ 
> Now I'm running into some different issues. The temperature and speed no longer change for the soluble material and the fan doesn't turn off, which makes the material come out bubbly.

Maybe you erroneously chose a wrong filament preset for the soluble support interfaces?

If you still have the problem, please provide me with the exported config file and an AMF. Thanks.

bubnikv, 2017-10-21:
There will be a patch release 1.37.2 released any minute, please give me a detailed feedback if there is still something fishy.

KeelinJ, 2017-10-21:
@bubnikv Thanks for the help. I did have the wrong material selected. Sorry for the trouble!

I'm printing again now, and both materials are printing well so far. I'll report back once the print has completed (it will take several hours).

bubnikv, 2017-10-22:
@KeelinJ Great. I have already released the patch version. Let's see whether this will be the last patch version of this series, or whether I will have to amend it once again :-)

bubnikv, 2017-10-22:
https://github.com/prusa3d/Slic3r/releases/tag/version_1.37.2

bubnikv, 2017-11-06:
Fixed with 1.37.2 release.

closed 2017-11-06 by bubnikv

KeelinJ, 2017-11-06:
@bubnikv Works great! Thanks for the fix

MStylesMS, 2017-11-06:
I will test later today.

Mark Stevens
[cid:E99B5747-1535-4613-8A49-0567F764ABEB]