prusa3d/PrusaSlicer #493: [OSX] Window header stuck behind menu bar

chamnit, 2017-08-27:
### Version
1.34.1.24-prusa3d

### Operating system type + version
Mac OSX 10.12.6

### Behavior
The Slic3r window title bar is stuck behind the OS X menu bar, along the top of the screen, as shown in the picture. The window can't be moved, and the three window buttons are hidden so it can't be closed, minimized, or maximized. The window can be only resized from the bottom and sides due to still being accessible. Slic3r has been stuck this way for several weeks and rebooting does not fix the problem.

[image]

It's unclear how the window got stuck this way, but there seems to be no direct way to fix this problem that I can see. Altering the screen resolution does move the window, uncovering the title bar and allows it to be manipulated, but changing the screen resolution back doesn't save its new position. It remains stuck behind the main menu bar.

I've had similar things happen in the past with other OS X programs. To fix the problem, I would select the 'Window' drop down in the main menu bar and use the common built-in 'Zoom' function. This would usually reposition the window so it could be seem in totality and immediately fix the problem. There is no 'Zoom' option in Slic3r's 'Window' drop down menu.

Probably the easiest way to fix this is to add the typical 'Zoom' and 'Minimize' commands in the 'Windows' drop down menu. Unless I am missing something totally obvious.

_Is this a new feature request?_ No

bubnikv, 2017-08-28:
Could you reproduce this issue? How often does it happen?

chamnit, 2017-08-28:
No. Like I said, I don't know how it happened and it has been stuck this way for weeks.

chamnit, 2017-09-06:
@bubnikv : I found a work around. I sometimes use a remote desktop service that automatically resizes the local screen based on what kind of device is connecting. I connected today and the window bar appeared in one particular instance. So, I was finally able to drag and move the window away from the top. FWIW, I did try resizing the screen in the Mac display settings and it didn't expose the menu bar.

If I were to guess, this remote desktop service was the original cause of the problem. I may have used Slic3r remotely to get a print job set up. So, this issue can probably be categorized as rare, but the suggested solution would still be the same. Make the traditional 'Zoom' function available in the 'Window' drop down menu.

bubnikv, 2019-01-28:
It will be finally fixed in 1.42.0-alpha4. Hurray!
Closing

closed 2019-01-28 by bubnikv