prusa3d/PrusaSlicer #494: Ubuntu 16.04 crashes on preset save

hexane360, 2017-08-27:
### Version
version_1.37.0-27-g3469bb2

### Operating system type + version
Ubuntu 16.04 LTS

### Behavior
Whenever I save a preset, Prusa slic3r immediately crashes. By running with stack trace using `perl -d Slic3r.pl`, I was able to get the following:
```
Slic3r::GUI::Tab::save_preset(/home/colin/Documents/3d/Slic3r-prusa/lib/Slic3r/GUI/Tab.pm:193):
193:	    $self->select_preset_by_name($name);
Slic3r::GUI::Tab::select_preset_by_name(/home/colin/Documents/3d/Slic3r-prusa/lib/Slic3r/GUI/Tab.pm:257):
257:	    my ($self, $name) = @_;
Slic3r::GUI::Tab::select_preset_by_name(/home/colin/Documents/3d/Slic3r-prusa/lib/Slic3r/GUI/Tab.pm:259):
259:	    $name = Slic3r::normalize_utf8_nfc($name);
terminate called after throwing an instance of 'std::bad_cast'
  what():  std::bad_cast
Aborted (core dumped)
```
Despite crashing, the config still saves properly and can be loaded next run. This happens with any configuration, new or old, Print Filament or Printer.

_Is this a new feature request?_ No

bubnikv, 2017-09-12:
Is it still crashing with the latest source?

hexane360, 2017-09-13:
It appears to be fixed now. Thanks.

closed 2017-09-13 by hexane360

bubnikv, 2017-09-13:
thanks for the report.