prusa3d/PrusaSlicer #495: Modify Variable Layer Heights To Use Print and Filament Settings Per Z Range

michmerr, 2017-08-27:
### Version
1.36.02-prusa3d-win64

### Operating system type + version
Windows 10

### Behavior
**Feature Request**

Setting variable layer height overrides allows only for the layer height to be changed. This does not allow for the inclusion of any other print characteristics that are often associated with printing with different layer heights, whether that be speeds, solid infill layers at the top/bottom, etc.

Current behavior: 
- From the _Plater_ tab:
 - Click the _Settings_ button.
 - Select the _Layers_ tab.  
 - **Set layer height overrides for defined Z ranges.**

Desired behavior:
- From the _Platter_ tab:
 - Click the _Settings_ button.
 - Select the _Layers_ tab.  
 - **Set _Print_ and _Filament_ settings overrides for defined Z ranges.**

Or (if we're feeling ambitious), add a view in _Plater_ with sliders for Z ranges that is independent of the current layer heights, and allow the _Print_ and _Filament_ settings selected using the existing controls in the current _Plater_ tab view to be applied to the selected Z range.

### Comments
(I originally added this as a comment to #378, but it looks like the focus of that feature request is no longer on what I am describing here.)

I'm using this feature for the first time, and before I found the dialog to assign layer height overrides, I was looking for something like the layer range sliders in the print preview, with the ability to set the Print Settings (and possibly Filament) for the selected layer range. In my scenario, I wanted an initial, thinly sliced region to use more solid infill layers on the top and bottom, but wanted fewer infill layers in the range using thicker layers. This is a simple scenario; single extruder, no filament swapping, just a desire to apply all aspects of a Print Setting to the layer range.

It seems like assigning settings to layer ranges, the same way we now assign them to the entire job, would allow for more detailed customization during layer editing, using existing, recognizable configuration structures.

bubnikv, 2017-08-28:
As of now, you can achieve the same effect using the modifier meshes. You
may create a "slab" modifier mesh in the Slic3r. I know it could be made
more convenient, but the tools are there already.

konvoj, 2017-09-06:
This is basically adding of the Processes as in Simplify3D. This is one of the last features that are still missing in other slicers.
Yes the modifiers are there, but it is way much easier to work with Processes. I would be very happy to see this in Slic3r as well.

bubnikv, 2018-08-02:
@YuSanka is revamping the UI. We have discussed the notion of modifier attributes per layer spans.

lukasmatena, 2026-09-01:
This is more or less implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena