prusa3d/PrusaSlicer #496: Crash when performing Cut/Keep Upper Part

BirkBinnard, 2017-08-28:
[binnard.zip](https://github.com/prusa3d/Slic3r/files/1257251/binnard.zip)
### Version
1.36.2

### Operating system type + version
Win10-64

### Behavior
Program crashes attempting to keep upper part of cut @ Z = 180. Error message says "..References memory at 0x0000000000000000000008. The memory coiuld not be read."

#### STL/Config (.ZIP) where problem occurs
binnard.zip contains STL + config bundle

Hugoz12, 2017-08-28:
Same with 1.35.5 linux64.

bubnikv, 2017-08-28:
Isn't it a duplicate of https://github.com/prusa3d/Slic3r/issues/105 ?

BirkBinnard, 2017-08-28:
Yes, I think this is a dupe of #105. Seems like overlapping shells is the factor that triggers it. I have done cuts OK before - but not with an STL with overlapping shells. Looks like I'll just have to print the whole thing and see what happens.  

I have printed parts with overlapping shells before, so the GCode Plic3r produces in these instances is OK. But for this new part I wanted to do a test of just the tippy top to make sure it would print OK. 

Oh well - at least Plic3r slices OK and produces valid GCode.  Neither S3D nor Craftware will do that.

closed 2017-08-28 by BirkBinnard

Hugoz12, 2017-08-29:
With OpenSCAD I managed to split it and then export it into a STL again. Have fun with it. :)
[2twistlamp3.zip](https://github.com/prusa3d/Slic3r/files/1261036/2twistlamp3.zip)

BirkBinnard, 2017-08-29:
Hey - thanks a bunch.  I'm printing the top part now.  I'll have to take a look at OpenSCad - so far I only use Rhino & Grasshopper.