prusa3d/PrusaSlicer #497: multiple perimeters on multipart object with single extruder

jindrichbenes, 2017-08-29:
Multiple perimeters are created when printing multipart object with single extruder, if multipart object has assigned different extruders.

[image]
single extruder, multipart object with extruders 1-4 assigned

[image]
same object, 3 parts set to default extruder, one to extruder 4

[image]
same object, all parts set to default extruder

bubnikv, 2017-11-06:
Would you please attach an amf file and a config? Thanks.

jindrichbenes, 2017-11-07:
[Desktop.zip](https://github.com/prusa3d/Slic3r/files/1449796/Desktop.zip)

bubnikv, 2018-08-02:
Current slic3r behaves differently, it will not allow one to slice object with non-existing extruders assigned.
In the future we may allow printing multi-material prints on single-extruder machines by ignoring the extruder IDs, but closing for now.

closed 2018-08-02 by bubnikv