prusa3d/PrusaSlicer #499: Per Filament Layer change G-Code

webhive, 2017-08-31:
New feature

It could be great to have a per filament layer change G code. 

I am using delay command for Layer change custom G-code to cool down filament and so harden it. Without delay on small parts cooler is not enough and better wait a little while filament became hard. 
But this approch is very filament specific - some filament hardened well without such trick, but some - not. 
   
So now using it via Printer Layer change code and this is not really useful. Way better to keep this setting separate for each filament.

Anyway not an urgent thing, but suppose it is nothing extra hard to add such featire.

bubnikv, 2017-09-12:
> I am using delay command for Layer change custom G-code to cool down filament and so harden it. 

For what purpose are you doing the wait? To cool down the print or the filament? Isn't the auto cooling feature sufficient?

webhive, 2017-09-12:
Cool down filament. Got situation when cooling is not enough on small parts. Really may be this is a matter of more advanced cooling nozzle. But with my current one (and sure I am not the only one with such nozzle) cooling is not enough, so using delay to cool down filament and it really help a lot.

quirxi, 2017-10-26:
You could use the "Before Layer change G-code" in the printer settings. I know its not the same as on a per filament basis, but it would work at least:

[image]

webhive, 2017-10-26:
@quirxi I know it. Using it now. It work but more logical to put it to per filament feature.

bubnikv, 2017-12-05:
If your setup is a "single heater multiple filaments" kind, then the upcoming Slic3r PE release will emit the filament_start_gcode after each tool change and filament_end_gcode before each tool change.

In any case, it will emit the toolchange_gcode at each tool change. In the toolchange_gcode you may read out the previous_extruder and next_extruder integer variables. You may use those variables in conditional expressions, like:
{if next_extruder==0} emit something {elsif next_extruder==1} emit something else {else} yet something else {endif}

Does this satisfy your needs?

webhive, 2017-12-05:
Is this filament base setting? Mean can I setup it for each filament differently and do not adjust it on every print?

bubnikv, 2017-12-05:
Sorry, I don't understand the question.

webhive, 2017-12-05:
> You may use those variables in conditional expressions, like:
{if next_extruder==0} emit something {elsif next_extruder==1} emit something else {else} yet something else {endif}

Is this per filament setting? Mean can I write these expression for each filament separately?

bubnikv, 2017-12-05:
You may place the conditional expression into any custom G-code section.

For example, we control the filament specific linear advance value from the
filament_start_gcode based on the active printer profile with the following
expression:
M900 K{if printer_preset=="Original Prusa i3 MK2 MultiMaterial"}200{elsif
printer_preset=="Original Prusa i3 MK2 MM Single Mode"}200{else}10{endif};
Filament gcode

webhive, 2017-12-05:
> You may place the conditional expression into any custom G-code section.

And update it for each print? I am trying to get rig of it. Currently Slic3r PE has no filament specific section, so I am unable to to what I need. So for each print if my filament changed I need to update this code. This is a reason why I asked for this feature.

So if I have currently about 10 filaments you say I should write huge `if elsif elsif ...` monster. No thank you - this is not a variant. It can be suitable for different printers (I dont think lot of printers is a frequent situation), but for case of filament it is real pain to keep this code updated.

bubnikv, 2017-12-05:
Slic3r PE has a filament specific start and end G-code sections. These sections are applied at the start and end of the print if the single_extruder_multi_material option is off, or every time a filament is switched if the single_extruder_multi_material is on.

bubnikv, 2018-08-02:
> So if I have currently about 10 filaments you say I should write huge if elsif elsif ... monster. No thank you - this is not a variant. It can be suitable for different printers (I dont think lot of [...]

You can use the "filament_type" field to simplify the huge if/elsif chain. You can be quite creative with the conditional g-code.

https://github.com/prusa3d/Slic3r/wiki/Slic3r-Prusa-Edition-Macro-Language
https://github.com/prusa3d/Slic3r/wiki/Slic3r-placeholders-(a-copy-of-the-mauk.cc-page)

also in the upcoming Slic3r 1.41.0 beta, you will find the name of the variable at the bottom of the parameter tooltip.

Closing.

closed 2018-08-02 by bubnikv