prusa3d/PrusaSlicer #500: Compilation error on make

Kuro-Maii, 2017-09-01:
### Version
git version: 247070cd8260d174162bb0cf21ead7d69af77f53

### Operating system type + version
Ubuntu 16.10
$ uname -a
Linux Yggdrasil 4.8.0-59-generic #64-Ubuntu SMP Thu Jun 29 19:38:34 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

### Behavior
* Make fails to compile
* _Steps needed to reproduce the problem:_
  * $ perl Build.PL # succes
  * $ perl Build.PL --gui # succes
  * ( output tells to change to build dir and follow the steps to run make )
  * $ cmake .. -DCMAKE_BUILD_TYPE=Release # succes
  * $ make # fail!
* _Expected Results:_ Successful Compilation
* _Actual Results:_
  ```
$ make
[  2%] Built target polypartition
[  4%] Built target clipper
[  8%] Built target libslic3r_gui
[  9%] Building CXX object xs/CMakeFiles/libslic3r.dir/src/libslic3r/ExtrusionSimulator.cpp.o
In file included from /usr/include/boost/config.hpp:61:0,
                 from /usr/include/boost/mpl/aux_/config/msvc.hpp:19,
                 from /usr/include/boost/mpl/aux_/config/adl.hpp:17,
                 from /usr/include/boost/mpl/aux_/adl_barrier.hpp:17,
                 from /usr/include/boost/mpl/bool_fwd.hpp:17,
                 from /usr/include/boost/mpl/bool.hpp:17,
                 from /usr/include/boost/mpl/not.hpp:17,
                 from /usr/include/boost/mpl/assert.hpp:17,
                 from /usr/include/boost/geometry/core/closure.hpp:22,
                 from /usr/include/boost/geometry/geometry.hpp:25,
                 from /usr/include/boost/geometry.hpp:17,
                 from /home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp:9:
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_half<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:277:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(half, 5.000000000000000000000000000000000000e-01, "5.00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:277:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_third<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:278:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(third, 3.333333333333333333333333333333333333e-01, "3.33333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333333e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:278:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_twothirds<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:279:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(twothirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666667e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:279:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_two_thirds<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:280:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(two_thirds, 6.666666666666666666666666666666666666e-01, "6.66666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666666667e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:280:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_three_quarters<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:281:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(three_quarters, 7.500000000000000000000000000000000000e-01, "7.50000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000e-01")
   ^
/usr/include/boost/math/constants/constants.hpp:281:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_two<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:282:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_two, 1.414213562373095048801688724209698078e+00, "1.41421356237309504880168872420969807856967187537694807317667973799073247846210703885038753432764157273501384623e+00")
   ^
/usr/include/boost/math/constants/constants.hpp:282:3: note: use -std=gnu++11 or -fext-numeric-literals to enable more built-in suffixes
/usr/include/boost/math/constants/constants.hpp: In static member function ‘static constexpr T boost::math::constants::detail::constant_root_three<T>::get(const mpl_::int_<5>&)’:
/usr/include/boost/math/constants/constants.hpp:283:3: error: unable to find numeric literal operator ‘operator""Q’
   BOOST_DEFINE_MATH_CONSTANT(root_three, 1.732050807568877293527446341505872366e+00, "1.73205080756887729352744634150587236694280525381038062805580697945193301690880003708114618675724857567562614142e+00")
   ^
[... 401 lines cut ...]
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp: In member function ‘void Slic3r::ExtrusionSimulator::set_image_size(const Slic3r::Point&)’:
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp:908:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
  for (size_t r = 0; r < image_size.y; ++ r) {
                     ~~^~~~~~~~~~~~~~
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp:909:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
   for (size_t c = 0; c < image_size.x; c += 2) {
                      ~~^~~~~~~~~~~~~~
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp: In member function ‘void Slic3r::ExtrusionSimulator::extrude_to_accumulator(const Slic3r::ExtrusionPath&, const Slic3r::Point&, Slic3r::ExtrusionSimulationType)’:
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp:961:8: warning: unused variable ‘h’ [-Wunused-variable]
  float h = scale_(path.height) * scalex;
        ^
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp: In member function ‘void Slic3r::ExtrusionSimulator::evaluate_accumulator(Slic3r::ExtrusionSimulationType)’:
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp:996:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int j = 0; j < pimpl->bitmap_oversampled; ++ j) {
                     ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/angel/projects/others/Slic3r-PE/xs/src/libslic3r/ExtrusionSimulator.cpp:997:24: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
      for (int i = 0; i < pimpl->bitmap_oversampled; ++ i) {
                      ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~
xs/CMakeFiles/libslic3r.dir/build.make:302: recipe for target 'xs/CMakeFiles/libslic3r.dir/src/libslic3r/ExtrusionSimulator.cpp.o' failed
make[2]: *** [xs/CMakeFiles/libslic3r.dir/src/libslic3r/ExtrusionSimulator.cpp.o] Error 1
CMakeFiles/Makefile2:222: recipe for target 'xs/CMakeFiles/libslic3r.dir/all' failed
make[1]: *** [xs/CMakeFiles/libslic3r.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
```

Kuro-Maii, 2017-09-01:
I just checked if I was missing something from libboost.
apt reports everything to be up to date.
I am using boost-1.61-dev

Kuro-Maii, 2017-09-04:
I am under the impression that most errors are due to different dependencies from the original slic3r by alexrj.
if this is the case then I advice the maintainers to link the installation instructions and all other files to this repo instead of the one from alexjr.
and also update the readme to clearly state the dependencies that this flavor of slic3r depends on.
for now I did find 2 dependencies and installed them.
however, this did not resolve the issue at hand.

agrill, 2017-09-10:
Same problem here on Ubuntu 17.04 on master.
Tried against libboost1.61-dev, libboost1.62-dev and libboost1.63-dev.

agrill, 2017-09-11:
Seems to be only a configuration issue in cmake. Adding the compiler flag -fext-numeric-literals (as suggested in the errors) solves the Issue.

bubnikv, 2017-09-11:
Would you please test that the issue has been fixed?
https://github.com/prusa3d/Slic3r/commit/836dd98113c10d532af77db89c6b77ac26c4c25e

I consider this issue to be a problem of Boost, which incorrectly uses GCC extensions in case the GCC extensions are not enabled. Unfortunately it worked on my systems.

agrill, 2017-09-11:
Yes, builds for me now on a clean working dir.

bubnikv, 2017-09-12:
thanks to all involved. Closing.

closed 2017-09-12 by bubnikv

Kuro-Maii, 2017-09-19:
Sadly no joy for me, the problem was not fixed.
I even recloned and checked out the commit specified.
so for the commit mentioned here this final output was generated with the following command:
`perl Build.PL && perl Build.PL --gui && take BUILD && cmake .. -DCMAKE_BUILD_TYPE=Release && make | pastebinit`
http://paste.ubuntu.com/25573186/

I also tested with the current master commit: 473233019cf6b3d0e6a9aac5b4b8e1b904e5d9ee and ran the same command as above when it was clear where the whole compile chain failed:
http://paste.ubuntu.com/25573201/

as a rule of thumb, from me as a developer, I would like to suggest to makes sure all compilations treat any warning as an error. this to prevent clutter in the output when somebody has to send in the out out when an error occurs. and it will also look nicer for the end user.
second, from me as an end user, please update the readme and link to installation instructions for this repo. the differences in compiling and dependencies between this and the linked repo from alexrj are too big to depend on their wiki and instruction.

greetings,
Kuro

bubnikv, 2017-09-19:
This is a new issue, it compiles on Windows, but not on Linux.

Kuro-Maii, 2017-09-19:
shall I copy paste it to a new issue then?

bubnikv, 2017-09-19:
This is a new issue, I will likely fix it tomorrow morning.

I recommend you to compile the latest release.

Kuro-Maii, 2017-09-19:
what would "the latest release" be?
the stable branche does not compile either.

bubnikv, 2017-09-20:
These are the releases:
https://github.com/prusa3d/Slic3r/releases