prusa3d/PrusaSlicer #501: [Request] Filament Shaping Retraction

sarf2k4, 2017-09-02:
Hi, I encountered filament jamming when I am printing with specific PLA on my prusa MMU printer. Latest was my second jam issues and I cranked up the hot end to 280'c so that I'm able to pull it out using unload filament menu and it worked.

But I wonder, since unloading filament and toolchange will be used in the future, the filament will be shaped so that there are no huge blob at the end of the filament end that was inside the hotend side thus providing smooth transition.

What if Filament Shaping Retraction feature were to be incorporated into this to avoid filament jamming for the famous PLA issues? It might be that Prusa had tested this but I doubt it. If there were tests regarding this in Prusa Research center I would like to hear the thoughts on this. Even the up-to-spec 2.0-2.1mm heatbreak also had some PLA jamming.

bubnikv, 2017-09-12:
> What if Filament Shaping Retraction feature were to be incorporated into this to avoid filament jamming for the famous PLA issues?

Frankly I don't understand your request. Would you please provide us with more details?

> What if Filament Shaping Retraction feature were to be incorporated into this 

into what?

sarf2k4, 2017-09-12:
When we used the "unload filament" from the menu, the printer will shape
the end of the filament to avoid huge blobs by stroking the filament in and
out several time.

Correct me if i am wrong here. From what i understand on the current
retraction, slic3r usually just instructs the printer to pull the filament
but there were no stroking movement, hence a blob might be formed. So the
"stroking" movement of the "unload filament", added into the normal
retraction. For example extruder retract value is 6mm, within that 6mm, it
will be divided into smaller segments of "push" and "pull" the filament to
avoid the blob being formed.

I myself unsure if this helps with the jam issues with pla but i am finding
some solutions to the jamming issues by asking whether prusa team tried
this method yet?

bubnikv, 2017-09-13:
> When we used the "unload filament" from the menu, the printer will shape
the end of the filament to avoid huge blobs by stroking the filament in and
out several time.

The procedure is a bit more complex. It is not required during a normal
retract as long as the end of the filament stays close to the melt zone,
that means during the normal print. But the filament shaping procedure is
required, if you are up to pulling the filament out of the printer
completely, or to swap the filaments.

> I myself unsure if this helps with the jam issues with pla but i am
finding
some solutions to the jamming issues by asking whether prusa team tried
this method yet?

The "filament end shaping procedure" is highly complex and it is material
specific. We had some of it in some firmware versions, but it was
simplistic and not working well.

We are considering to add the "filament end shaping" procedure into the
firmware, but it is quite a lot of work as it is filament specific and
there are differences between pushing the filament into the wipe tower and
pushing it into a free air.

Anyway, the firmware issue tracker is likely a better place to discuss this
topic.
https://github.com/prusa3d/prusa-firmware/issues

sarf2k4, 2017-09-13:
I started this discussion because this might benefits on other printers as well who are using all metal type hotend. 

I am rooting on this one

bubnikv, 2017-11-06:
Actually it may be more sensible to let Slic3r generate a "cancelation" G-code block and mark it with some label, and to instruct the firmware to jump to that label on print abort. This  is a similar feature to https://github.com/prusa3d/Slic3r/issues/538

bwright86, 2017-11-19:
Not sure if this helps with your issue, but I have played with a start and stop g-code sequence in Prusa's Slic3r. It started with a post from "PJR" in the Prusa3d forums, and morphed into a helpful way to start and end print jobs.

To explain what is being done. The start code gets the extruder up to 180 degrees, and starts heating the bed, then performs the mesh bed leveling calibration. Afterwards it raises the Z-axis to 20, and waits for both extruder/bed to reach the desired temp, then starts restoring filament into the extrusion chamber. Finally it runs through and spits out some filament to prepare for the print.

The End gcode runs through the process to reform the tip, and backs the filament out. This allows you to start every print job with no filament in the extruder, and is ready for any extruder to start a print job. It also includes PJR's tip reforming formula ([Link](https://shop.prusa3d.com/forum/original-prusa-i3-mk2-multi-material-f26/retraction-procedure-t4868.html#p35493)).

**Start G-code**
```
M115 U3.0.12 ; tell printer latest fw version
; BEGIN Start G-code sequence
T[initial_tool]
M140 S[first_layer_bed_temperature]     ; set bed temp
M109 S180    ; wait for extruder temp to reach 180 degrees.
G21          ; set units to millimeters
G90          ; use absolute coordinates
M83          ; use relative distances for extrusion
G28 W        ; Home all Axis
G80          ; Automatic mesh bed leveling
G1 Z20 F800  ; Raise Z-Axis while heating
M104 S[first_layer_temperature]         ; set extruder temp
M190 S[first_layer_bed_temperature]     ; wait for bed temp
M109 S[first_layer_temperature]         ; wait for extruder temp
G92 E0.0                                ; Set current position to coordinates given

M203 E100
M92 E140                     ; Set axis_steps_per_unit
G1 Z0.250 F7200.000
G1 E85 F500                 ; Restore filament into extruder
G1 Z0 F800                  ; Lower Z-Axis after heating
G1 X50.0 E30.0  F1000.0
G1 X160.0 E20.0  F1000.0
G1 Z0.200 F7200.000
G1 X220.0 E13 F1000.0
G1 X240.0 E0 F1000.0
G1 E-4 F1000.0
G92 E0.0
; END Start G-code sequence
```

**End G-code**
; BEGIN End G-code sequence
G91                              ; Use relative coordinates
G1 Z10.0 F7200	; Lift Z and retract
G1 X0 Y200 F3600		; present finished print job
G1 Z60 F2000 		; Set Z above the print bed for next job
; BEGIN filament unload
; Retract filament 92mm
G1 E12 F210	; unload filament
G1 E-80 F5000	; bring filament to main steel tube
; BEGIN Reform tip
G1 E-5 F3500
G1 E-5 F2400
G1 E-2 F1000
G1 E10 F300
G1 E-10 F300
G1 E10 F600
G1 E-10 F600
G1 E10 F900
G1 E-10 F900
G1 E15 F1200
G1 E-15 F1200
G1 E5 F600
; END Reform tip
; END filament unload
G4 S1	; pause 1 second
M107	; turn off fan
M104 S0 	; turn off temperature
M140 S0 	; turn off heatbed
M84	; disable motors
; END End G-code sequence

I have a seperate job that will do the same end G-code sequence, and I run it if a print job is cancelled.

This profile has worked well for ABS and PETG, I have not really experimented on it with PLA.

bubnikv, 2017-11-28:
I believe we have similar start / end G-code sequences defined in our Slic3r settings "Original Prusa i3 MK2 MM Single Mode" as Peter has. Currently the unload procedure defined for the "Original Prusa i3 MK2 MM Single Mode" profile contains the finalization sequence normally generated for the wipe tower for the PLA filament. We plan to make use of the new conditional G-codes (to be released with the Slic3r 1.38.xx) and we will make the unload sequence dependent on the filament type.

To give you an idea, in Slic3r 1.38.1 one will be able to place conditional blocks like this:

{if filament_type=="PLA"}
do something for PLA
{elsif filament_type=="ABS"}
do something for ABS
{else}
do the default thing
{endif}

sarf2k4, 2017-11-28:
That is interesting, i presume that the conditional checking were done by
slic3r right?

With the conditionals, this will give even more play thing in the g-code
generation especially for the "per filament g-code" script for the linear
advance K value on different printers

bubnikv, 2017-12-05:
> I presume that the conditional checking were done by slic3r right?

Yes, they are already in Slic3r 1.38.3 beta. The error reporting in 1.38.3 on syntax error in the conditional syntax is lacking, it will be improved in the next release.

The examples of the conditional syntax for the linear advance are to be found in our filament profiles:
https://github.com/prusa3d/slic3r-settings/tree/merged

May I please close this issue?

bubnikv, 2018-02-12:
No response, closing this issue.

closed 2018-02-12 by bubnikv