prusa3d/PrusaSlicer #502: [FeatureRequest] Make Infill/Support Setting selectable from the platter menu

ManuGithubSteam, 2017-09-04:
### Version
1.36.2
_Use `About->About Slic3r` for release versions_

### Operating system type + version

### Behavior
* _Describe the problem_

If i want to change Infill/ Set supports of a part i have to click trough the menus and sometimes forget was setting i choose last time.

* _Expected Results_

Make a menu in the platter tab like the cura one with the most used infill settings.

I would suggest: 0% 10% 15%

See Screenshot mockup :-)

[image]

numanair, 2017-09-10:
I think there is a better way to approach the UX than the way overrides are done in the original Slic3r. When I used it I did not notice they were there at first and they were causing old settings to stick.

bubnikv, 2017-10-03:
My impression of the overrides is, it tries to copy the new Cura interface. I am not a big fan of both either. I think we need to go in the direction of the Prusa Control UI for the basic settings and then to have the good old Slic3r parameter pages for the fine adjustments. It takes a development effort to implement that though.

bubnikv, 2018-03-14:
The following new controls are checked into the current master, they will make it into the Slic3r PE 1.40.

[image]

closed 2018-03-14 by bubnikv

ManuGithubSteam, 2018-05-26:
Yes this is great!
I would love to see another drop down menu to select the type of infill :-)