prusa3d/PrusaSlicer #503: Perimeter shells delaminate

natewalck, 2017-09-04:
### Version
1.36.2

### Operating system type + version
macOS 10.12.6

### Behavior
* When printing a model with rectangular features, the perimeter shells dellaminate (As seen here: 
* Print 3D Benchy (or any other model with rectangular features) with 3 vertical shell perimeters. 
* Should have a solid perimeter
* [3D Benchy Fail](https://i.imgur.com/WFnvTRG.jpg)

#### STL/Config (.ZIP) where problem occurs
[BenchyFail.zip](https://github.com/prusa3d/Slic3r/files/1275507/BenchyFail.zip)

bubnikv, 2017-10-03:
This is really an under-extrusion issue, caused by an incorrect extruder calibration, by low extruder temperature, by an excessive print speed or by a partially clogged nozzle. I don't think this is a Slic3r issue.

Closing.

bubnikv, 2017-10-03:
You may also need to check your pulleys. I have actually seen a similar issue myself on my printer, where one of the pulleys has not been tightened enough against its motor axis, causing a slack in the axis movement.

closed 2017-10-03 by bubnikv