prusa3d/PrusaSlicer #505: White spaces in path under Linux

diego191, 2017-09-06:
### Version
1.36.2-prusa3d-linux64

### Operating system type + version
Linux Mint 18.2 Sonya

### Behavior
It's not possible to import stl in console with white spaces in path. For example:
```
slic3r /home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl
slic3r "/home/diego/CEL Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl"
slic3r '/home/diego/CEL Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl'

```
Everytime the slic3r says:

```
Input file must have .stl, .obj or .amf(.xml) extension
Can't call method "objects_count" on an undefined value at /tmp/.mount_slic3rViFVgx/usr/bin/lib/site_perl/5.22.0/Slic3r/Model.pm line 17.
Multiple objects were loaded for a multi-material printer.
Instead of considering them as multiple objects, should I consider
these files to represent a single object having multiple parts?
```

In the command line can be seen following:
```
04:39:14 PM: Debug: Unrecognized accel key ' del', accel string ignored.
04:39:14 PM: Debug: Unknown accel modifier: ' '
04:39:14 PM: Debug: No accel key found, accel string ignored.
04:39:14 PM: Debug: Unknown accel modifier: ' '
04:39:14 PM: Debug: No accel key found, accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' l', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' r', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' s', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 0', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 1', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 2', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 3', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 4', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 5', accel string ignored.
04:39:14 PM: Debug: Unrecognized accel key ' 6', accel string ignored.
stl_initialize: Couldn't open Robox/PrintJobs/6b8e4945ab8741a2/6b8e4945ab8741a2-0.stl for reading: No such file or directory
```

### Workaround
```
cd /home/diego/CEL\ Robox/PrintJobs/6b8e4945ab8741a2
slic3r 6b8e4945ab8741a2-0.stl

```

bubnikv, 2017-10-03:
Where did you get the Slic3r from?

I just tested
Slic3r-1.37.1-prusa3d-linux64-full-201709141215.tar.bz2
from
https://github.com/prusa3d/Slic3r/releases
and it works with all the three ways of escaping the whitespaces you mentioned.

You will find a slic3r script with the following content in the tar.bz2 file:
#!/bin/bash
DIR=$(dirname "$0")
export LD_LIBRARY_PATH="$DIR/bin"
export PERL5LIB=
export PERLLIB=
exec "$DIR/bin/perl5.22.0" "$DIR/bin/slic3r.pl" "$@"

This script ensures that the paths with whitespaces are handled correctly.

I suspect that your slic3r script is different from the one above.

bubnikv, 2017-10-21:
No response, closing.
I hope this issue has already been fixed in the slic3r start script, it works in Slic3r-1.37.1.

closed 2017-10-21 by bubnikv