prusa3d/PrusaSlicer #506: Int128 is broken on RaspberryPi by a6ea01a

eyal0, 2017-09-07:
### Version
`a6ea01a`

### Operating system type + version
RaspberyPi running OctoPi

### Behavior
Unable to compile versions beyond `a6ea01a`.  The first file that I encountered that cannot be bulit is `FillRectilinear3.cpp`.  To reproduce on Raspberry Pi, check out that version and bulid.  The error log is below:

pi@octopi:~/Slic3r $ ./bisect.sh
/home/pi/Slic3r
```
cc -I/usr/lib/arm-linux-gnueabihf/perl/5.20/CORE -fPIC -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -xc++ -Isrc/libslic3r -Isrc/glew/include -Isrc/eigen -I/usr/lib/arm-linux-gnueabihf/wx/include/gtk2-unicode-3.0 -I/usr/include/wx-3.0 -D_GLIBCXX_USE_C99 -DHAS_BOOL -DNOGDI -DSLIC3RXS -DBOOST_ASIO_DISABLE_KQUEUE -DGLEW_STATIC -std=c++11 -DSLIC3R_GUI -DSLIC3R_PRUS -DUNICODE -D_FILE_OFFSET_BITS=64 -DWXUSINGDLL -D__WXGTK__ -pthread -DBOOST_LOG_DYN_LINK -DBOOST_LIBS -DNDEBUG -Isrc -Ibuildtmp -c -D_REENTRANT -D_GNU_SOURCE -DDEBIAN -fwrapv -fno-strict-aliasing -pipe -I/usr/local/include -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -O2 -g -o src/libslic3r/Fill/FillRectilinear3.o src/libslic3r/Fill/FillRectilinear3.cpp

src/libslic3r/Fill/FillRectilinear3.cpp:593:8: warning: extra tokens at end of #endif directive
 #endif _MSC_VER
        ^
In file included from src/libslic3r/Fill/FillRectilinear3.cpp:14:0:
src/libslic3r/Fill/../Int128.hpp:72:2: error: expected unqualified-id before ‘__int128’
  __int128 value;
  ^
src/libslic3r/Fill/../Int128.hpp: In constructor ‘Int128::Int128(int64_t)’:
src/libslic3r/Fill/../Int128.hpp:74:27: error: class ‘Int128’ does not have any field named ‘value’
  Int128(int64_t lo = 0) : value(lo) {}
                           ^
src/libslic3r/Fill/../Int128.hpp: In copy constructor ‘Int128::Int128(const Int128&)’:
src/libslic3r/Fill/../Int128.hpp:75:28: error: class ‘Int128’ does not have any field named ‘value’
  Int128(const Int128 &v) : value(v.value) {}
                            ^
src/libslic3r/Fill/../Int128.hpp:75:36: error: ‘const class Int128’ has no member named ‘value’
  Int128(const Int128 &v) : value(v.value) {}
                                    ^
src/libslic3r/Fill/../Int128.hpp: In member function ‘Int128& Int128::operator=(const int64_t&)’:
src/libslic3r/Fill/../Int128.hpp:77:42: error: ‘value’ was not declared in this scope
  Int128& operator=(const int64_t &rhs) { value = rhs; return *this; }
                                          ^
src/libslic3r/Fill/../Int128.hpp:77:42: note: suggested alternative:
In file included from /usr/include/boost/bind.hpp:22:0,
                 from /usr/include/boost/thread/detail/thread.hpp:29,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from src/libslic3r/libslic3r.h:14,
                 from src/libslic3r/Fill/../ClipperUtils.hpp:4,
                 from src/libslic3r/Fill/FillRectilinear3.cpp:10:
/usr/include/boost/bind/bind.hpp:112:25: note:   ‘boost::_bi::value’
 template<class T> class value
                         ^
In file included from src/libslic3r/Fill/FillRectilinear3.cpp:14:0:
src/libslic3r/Fill/../Int128.hpp: In member function ‘uint64_t Int128::lo() const’:
src/libslic3r/Fill/../Int128.hpp:79:42: error: ‘value’ was not declared in this scope
  uint64_t lo()   const { return uint64_t(value); }
                                          ^
src/libslic3r/Fill/../Int128.hpp:79:42: note: suggested alternative:
In file included from /usr/include/boost/bind.hpp:22:0,
                 from /usr/include/boost/thread/detail/thread.hpp:29,
                 from /usr/include/boost/thread/thread_only.hpp:22,
                 from /usr/include/boost/thread/thread.hpp:12,
                 from /usr/include/boost/thread.hpp:13,
                 from src/libslic3r/libslic3r.h:14,
                 from src/libslic3r/Fill/../ClipperUtils.hpp:4,
                 from src/libslic3r/Fill/FillRectilinear3.cpp:10:
/usr/include/boost/bind/bind.hpp:112:25: note:   ‘boost::_bi::value’
 template<class T> class value
                         ^
In file included from src/libslic3r/Fill/FillRectilinear3.cpp:14:0:
src/libslic3r/Fill/../Int128.hpp: In member function ‘int64_t Int128::hi() const’:
src/libslic3r/Fill/../Int128.hpp:80:41: error: ‘value’ was not declared in this scope
  int64_t  hi()   const { return int64_t(value >> 64); }
                                         ^
[... 277 lines cut ...]
           ^
src/libslic3r/Fill/../Int128.hpp: In static member function ‘static int Int128::compare_rationals(int64_t, int64_t, int64_t, int64_t)’:
src/libslic3r/Fill/../Int128.hpp:111:3: error: expected primary-expression before ‘__int128’
   __int128 det = __int128(p1) * __int128(q2) - __int128(p2) * __int128(q1);
   ^
src/libslic3r/Fill/../Int128.hpp:112:12: error: ‘det’ was not declared in this scope
   return ((det > 0) - (det < 0)) * invert;
            ^
In file included from src/libslic3r/Fill/../Line.hpp:5:0,
                 from src/libslic3r/Fill/../Polygon.hpp:7,
                 from src/libslic3r/Fill/../ExPolygon.hpp:5,
                 from src/libslic3r/Fill/../ClipperUtils.hpp:6,
                 from src/libslic3r/Fill/FillRectilinear3.cpp:10:
src/libslic3r/Fill/../Point.hpp: In instantiation of ‘TO Slic3r::convert_to(const Slic3r::Point&) [with TO = Slic3r::Pointf]’:
src/libslic3r/Fill/FillRectilinear3.cpp:224:54:   required from here
src/libslic3r/Fill/../Point.hpp:245:94: error: dependent-name ‘TO:: coord_type’ is parsed as a non-type, but instantiation yields a type
 template<typename TO> inline TO convert_to(const Point &src) { return TO(TO::coord_type(src.x), TO::coord_type(src.y)); }
                                                                                              ^
src/libslic3r/Fill/../Point.hpp:245:94: note: say ‘typename TO:: coord_type’ if a type is meant
src/libslic3r/Fill/../Point.hpp:245:117: error: dependent-name ‘TO:: coord_type’ is parsed as a non-type, but instantiation yields a type
 template<typename TO> inline TO convert_to(const Point &src) { return TO(TO::coord_type(src.x), TO::coord_type(src.y)); }
                                                                                                                     ^
src/libslic3r/Fill/../Point.hpp:245:117: note: say ‘typename TO:: coord_type’ if a type is meant

```

eyal0, 2017-09-07:
Looks like HAS_INTRINSIC_128_TYPE is true but it should probably be false for RPi.  I mean, I can't image that the little RPi has a 128-bit type!

eyal0, 2017-09-08:
You can assign it to me, by the way, I'm looking at it already.

bubnikv, 2017-09-11:
merged, thanks.

closed 2017-09-11 by bubnikv