prusa3d/PrusaSlicer #507: Under extrusion when changing speed ?

porlock, 2017-09-08:
### Version
Slic3r-1.36.2-prusa3d-win64-full-201707281218

### Operating system type + version
Windows 10

### Behavior
Im constantly get under extrusion, in some parts it is repeatable every print at the same spot, i think it happens when speed change occurs, there is nothing special in this place i think, but it keep happening i can solve why. I try another slicing program (with similar settings) and it prints fine without any under extrusion in the same places that Slic3rPE does,  I attache a gcode, stl file, config file and 2 photos.

#### STL/Config (.ZIP) where problem occurs
[image]
[image]
[under.zip](https://github.com/prusa3d/Slic3r/files/1287395/under.zip)

Hugoz12, 2017-09-08:
With your setting to slowdown at 5s layer time this can't be the reason for it as that value isn't reached at that layer. (In fact that's only the case with the last layer.)
But I saw something interesting. Load the gcode to gcode.ws and view **layer 74**.
I think I saw this error before being reported here. The printer performs a travelmove for about 10s then goes to the next layer without printing anything. The oozing within these 10s might be enough to result in the here seen layer error.

porlock, 2017-09-08:
i think is the line 90 when external perimeter speed drops from 50 do 10mm/s, i change setting small perimeter in slic3r for less speed difference and it was printed better.

bubnikv, 2017-10-03:
I believe it is not the Slic3r issue, but your settings or machine issue.

We recommend to start from our settings.
https://github.com/prusa3d/slic3r-settings

I am closing this issue, because I don't think this is a software issue.

closed 2017-10-03 by bubnikv