prusa3d/PrusaSlicer #508: Gap Speed is used when Gap Fill is not zig-zag and is the same as other perimeters.

blacklion, 2017-09-08:
### Version
1.36.2-prusa34-win64

### Operating system type + version
Win'7 Ultimate, but I don't think it is relevant

### Behavior
Slic3r uses "Gap Speed" (very low, intended for zig-zag movements) even when "Gap" in question is straight line (for example, gap in thin wall which is slightly wider that integer multiply of extrusion width).  It slows down print unnecessary. Standalone straight line is simple case, but there are more complex ones: for example, square or circle. In all cases, when "gap filling" has exactly same shape as perimeters, it could be printed at full infill speed and not at ridiculously slow "gap fill " speed, as there are no any "zig-zag" movements.

Three samples attached: simple bar (line), box (square) and pipe (cylinder). When they are sliced with attached config, every of this models has thickness of five perimeters (passes) and inner (middle) pass is done at "gap speed", though these middle passes are not different from four other perimeters at same layer!

[unnecessary-gap-speed.zip](https://github.com/prusa3d/Slic3r/files/1288789/unnecessary-gap-speed.zip)

bubnikv, 2017-09-11:
Why don't you just increase the gap speed then?

blacklion, 2017-09-11:
Because there could be zig-zags in other parts of model. Attached examples are minimum test cases, and not real models, of course.

bubnikv, 2017-09-11:
> Because there could be zig-zags in other parts of model. Attached
examples are minimum test cases, and not real models, of course.

But this scenario is not specific to gap fill: You may have zig-zags and
long moves in perimeters or an infill.

What is specific to the gap fill is the rapidly changing extrusion rate,
when there is a change in the line width.

blacklion, 2017-09-11:
Ok, arguments about zig-zags are taken from documentation and source code comments :)
But it is not too relevant to this case: anyway, some gap fills looks like not as special gap fills, they don't contains neither zig-zags nor width change and could be done at normal inner-perimeter speed.

closed 2026-09-29 by prusaslicer-bot[bot]