prusa3d/PrusaSlicer #511: AppImage always enters GUI mode, regardless of command line arguments

wcraigtrader, 2017-09-10:
### Version
Slic3r-1.36.2-prusa3d-linux64-full-201707281217.AppImage

### Operating system type + version
Ubuntu 16.04.3 LTS

### Behavior
* _Describe the problem_
  * Download https://github.com/prusa3d/Slic3r/releases/download/version_1.36.2/Slic3r-1.36.2-prusa3d-linux64-full-201707281217.AppImage
  * chmod 755 /opt/Slic3r/Slic3r-1.36.2-prusa3d-linux64-full-201707281217.AppImage
  * Run with appropriate command line arguments:
    ```
    /opt/Slic3r/Slic3r-1.36.2-prusa3d-linux64-full-201707281217.AppImage \
        --load slic3r.ini --output test.gcode outpost-card-tray-1-box.stl
    05:08:46 PM: Debug: Unrecognized accel key ' del', accel string ignored.
    05:08:46 PM: Debug: Unknown accel modifier: ' '
    05:08:46 PM: Debug: No accel key found, accel string ignored.
    05:08:46 PM: Debug: Unknown accel modifier: ' '
    05:08:46 PM: Debug: No accel key found, accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' l', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' r', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' s', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 0', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 1', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 2', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 3', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 4', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 5', accel string ignored.
    05:08:46 PM: Debug: Unrecognized accel key ' 6', accel string ignored.
    Info about outpost-card-tray-1-box.stl:
      size:              x=125.395 y=89.998 z=33.333
      number of facets:  684
      number of shells:  1
      volume:            60220.344
      needed repair:     no
    ```
* _Expected Results_
Slic3r should have created `test.gcode`.

* _Actual Results_
Instead of slicing my STL file, the Slic3r GUI starts and loads my STL file.

_Is this a new feature request?_
No.

#### STL/Config (.ZIP) where problem occurs
_Upload a zipped copy of an STL and your config (`File -> Export Config`)_

bubnikv, 2017-09-12:
This has been done on purpose by @probonopd.

Simon, what was your rationale to force the --gui mode for the appimage?

probonopd, 2017-09-12:
The assumption is that most Slic3r AppImage users will open Slic3r from the menu, and use the GUI. However, the current implementation is not ideal.
* By default, Slic3r runs as a command line tool
* It needs a `--gui` argument to run with a GUI
* Upstream Slic3r has changed the logic, can launch the GUI by default unless `--no-gui` was specifically requested. @bubnikv maybe you want to adopt that change. Please see https://github.com/alexrj/Slic3r/issues/4111

@wcraigtrader in the meantime, you could download the AppImage, edit the AppRun file (remove `--gui`) and use `appimagetool` to repack the extracted directory into an AppImage.

bubnikv, 2017-11-30:
I have applied the change by @lordofhyphens. Thanks @probonopd  for pointing to that commit.
https://github.com/prusa3d/Slic3r/commit/488feb23352b07c27b07468408f1318dd745810f

bubnikv, 2017-12-01:
Will be fixed with 1.38.3. Closing.

closed 2017-12-01 by bubnikv