prusa3d/PrusaSlicer #512: MM purge suggestion

3d-gussner, 2017-09-11:
As the filament needs to purge during material change for a defined distance, somewhere between 8-20 cm (depending on the colour/material), wouldn't it be possible to purge it as infill and/or duplicates of the model?

Guess most people would be happy to have one mm print that has 100% infill and no/less waste compared to a 5% infill and more than 100%-?000% waste.
Also having multiple mm copies with up to 100% infill would be great compared to the purge block waste.
Or having multiple mm copies with bit more infill as normally needed.

**Problem:**
The infill may not be enough to purge the needed length in some models.

**Possible solution:**
If it would be possible to calculate the needed infill length for mm changes and compare it with the used settings for the actual print, it should be possible to...

1. Increase the infill ratio to match the needed purge lengths.
2. Duplicate the model so many times it finally fits the needs.
3. Combination of 1 and 2 if 1 isn't fit the needs.
4. Some models may have tiny details at the top, so infill and duplication will not be an option, why not purge the last tiny things in a purge bucket (has to be designed) and wipe the nozzle in the back? Where the last 60mm-70mm are reserved for the purge bucket and the wipe line?

I know it limits the build size, but the default purge block is 60mm and so it is nearly the same.
 Why the last 60mm-70mm in the back? Moving the x-axis and extruder from a high z position to the back and down would not collide with the print. After the wipe line moving it up and back to previous position would also not collide with the print . The hole extruder assembly x-axis and cables would be always behind the actual print. Disadvantage of it is that the printer would also have to move z up and down during the purge, which could add some print time.

Using very different materials, like PLA/ABS/PETG and soluble material, would need another solution. Maybe a combination of my suggestion for the materials doing fine together as infill and purge the soluble material in a bucket.
Thinking of the high costs of soluble material: I am not sure but I think Slic3r can/will minimize the use of soluble material where needed and use 'normal' filament to build the support base for it.

**2nd thought:**
Print useful predefined prints in the back where the colour is not important, which would need bit more reserved pace depending on the 'useful print'.

**3rd brain fart:**
Define a virtual extruder 5 which primary uses purge filament. So it would be possible to print something the user needs but don't care about the colour by choosing that virtual extruder. This applies only for filaments which can be combined.
For soluble material would still need a purge block or bucket.

These are just some thoughts and I don't know if that is even possible to implement it without TOO much work.

bubnikv, 2017-09-12:
> As the filament needs to purge during material change for a defined distance, somewhere between 8-20 cm (depending on the colour/material), wouldn't it be possible to purge it as infill and/or duplicates of the model?

Yes. This is a natural thing, which is already on my list for a long time. The list is long though.

> Define a virtual extruder 5 which primary uses purge filament. So it would be possible to print something the user needs but don't care about the colour by choosing that virtual extruder. This applies only for filaments which can be combined.

This is my plan. There is an extruder ID "0" used currently for the "don't care" extruder used for our partially soluble supports. My idea is to apply the same trick for the assignment of extruders to the objects. It sounds easy, but there is a lot of code shuffling to be done to make it work.

> Guess most people would be happy to have one mm print that has 100% infill and no/less waste compared to a 5% infill and more than 100%-?000% waste.

I think you will still need the purge tower for the filament switch sequences as they are timing critical. But the actual color purge may be extruded elsewhere.

> Increase the infill ratio to match the needed purge lengths.

This is an interesting proposal, but it is difficult to modulate the infill density per layer as this would lead to badly fused infill patterns between the layers. It may work, but it is likely a lot of work.

> Some models may have tiny details at the top, so infill and duplication will not be an option, why not purge the last tiny things in a purge bucket (has to be designed) and wipe the nozzle in the back?

Something like this may work for the color purge, but just letting the purged material to fall down may make a lot of mess and / or will require more volume than the compressed wipe  tower.

> Thinking of the high costs of soluble material: I am not sure but I think Slic3r can/will minimize the use of soluble material where needed and use 'normal' filament to build the support base for it.

This has been implemented already, at least to some extent:
https://www.youtube.com/watch?v=wc6ui17G9bk

bubnikv, 2018-07-21:
Wipe into infill / sacrificial object are implemented in Slic3r 1.41.0-alpha2.

We are now considering to implement wipe into supports, which is complicated by the fact, that the support is printed at heights different from the object layers.

3d-gussner, 2018-08-26:
Thanks for the implementation

closed 2018-08-26 by 3d-gussner