prusa3d/PrusaSlicer #513: Allow to specify Z resolution separately of minimum layer height

DrLex0, 2017-09-11:
### Version
1.36.2

### Operating system type + version
Any

### Feature request

A minimum and maximum can be specified in the extruder settings for variable layer height. The expected behavior according to the tooltip of the Min field, is that layer Z values will be quantized to the value of that field. (This is currently not the case: #473).

I think we need more configurability here. I would like to specify a minimum layer height that is much larger than my printer's Z resolution, without losing Z positioning accuracy.
In my case, the resolution is 0.0025mm. I never want to print at such fine layers because prints would take insanely long with no visible benefit. I would therefore rather specify a minimum of 0.05mm, yet I still want every layer to be positioned with 0.0025mm precision on the Z axis. Of course I could try not to go below 0.05mm in the layer editing tool, but there is no way to see what the actual values are, and I don't want to have to edit the curve in a range of [0.0025, 0.3] while I only need [0.05, 0.3].

There should be three fields: Min, Max, and Resolution. Layer heights should be allowed to vary between Min and Max, and the actual layer height must be rounded to Resolution before slicing the layer (in general, not only when using variable heights).
Mind that Slic3r currently seems to round all Z values to 0.001mm in G-code, so this would also need to be updated to allow higher resolutions.

DrLex0, 2017-10-03:
This is a duplicate of #202.

closed 2017-10-03 by DrLex0

quetzalcoatl, 2021-01-30:
@DrLex0 nice Z-resolution. What printer was that?

DrLex0, 2021-01-30:
@quetzalcoatl It's a 2016 FlashForge Creator Pro. Those are the official specs, I have never checked it but it seems plausible given that the Z positioning is done through a leadscrew driven by a stepper, so a single step would indeed only amount to a very tiny movement…