prusa3d/PrusaSlicer #514: How to run post processing scripts under Windows 10 pro 64bit ?

Scachi, 2017-09-12:
### Version
1.36.2-prusa3d-stable-win64

### Operating system type + version
Windows 10 Pro 64bit Version 1607

### Behavior
How do I run a post processing script under my windows version ?

Under "Print-Settings" I have added this to the Post-processing scripts entry:
`C:\Users\scach_000\Documents\3D-Print\Software\slic3r_my.pl`

This opens just a cmd.exe window where it will echo something back when I type something into it.
It won't create the backup file.
Do I have to adjust something in Windows to actually pipe the gcode file through this script ?

This is the script slic3r_my.pl:
```
#!/usr/bin/perl -i
#
# Remove all toolchange commands (and T# arguments) from gcode.

use strict;
use warnings;

$^I = '.bak';
# read stdin and any/all files passed as parameters one line at a time
while (<>) {
    if (not /^T[0-9]/) {
        s/\s*(T[0-9])//;
        print;
    } else {
    }
}
```

bubnikv, 2017-09-12:
You have to put the whole path to the perl interpreter there on Windows,
something like
C:\Program Files\Prusa3D\Slic3r\perl5.24.0.exe  yourpathto\slic3r_my.pl
To tell the truth, I did not ever try that myself. It may be a good idea to
do this kind of substitution inside Slic3r automatically, but on the other
side we would like to get rid of Perl from the Slic3r installation one day.

Scachi, 2017-09-12:
Putting the whole path in isn't doing any good.
 It looks like it tries to interpret it as one command in this case, resulting in the message "The configured post processing script is not executable: check permissions.
No matter how I put the path in there / or \ without whitespaces or escaped ws.

I was trying to to use a script to work around some problems with my parking extruder setup.
It looks like using T1 T0 in the start gcode works for me. T0 wasn't picking up the extruder without changing the tools one time or specifying the pickup position and pin action manually.

Not sure if you want to close this issue, as it looks it isn't working at all under win10 64bit but as you are planning to get rid of perl anyway...

bubnikv, 2017-10-03:
You may write a batch file with the following content:
pathtoyourperl\perl.exe pathtoyourscript\remove.pl %*
and then you need to give Slic3r the full path to your batch file.

I have also changed Slic3r to automatically pick the perl interpreter on Windows to run the user provided perl script. If you manually merge this change with your slic3r installation, you may just provide the full path to your perl script to Slic3r.
https://github.com/prusa3d/Slic3r/commit/5a844c91f12ceb0fab87ec888f463d057bffdd06

bubnikv, 2017-10-21:
Closed with the release of Slic3r PE 1.37.2

closed 2017-10-21 by bubnikv