prusa3d/PrusaSlicer #515: Incorrect internal dimensions.

KostinPavel, 2017-09-12:
### Version
1.36.2-prusa3d-win64

### Operating system type + version
Windows 10 Pro 1703

### Behavior
* _Describe the problem_
Incorrect internal dimensions.
Internal dimensions less than the dimensions of the model to the width of the extrusion.

* _Steps needed to reproduce the problem_
Ideally:
We need to shift the internal perimeters inside the width of the extrusion.
Workaround:
We need a separate setting for independent compensation of the dimensions in the XY-plane for the inner and outer perimeters.

* _Expected Results_
[image]
[InputModel.zip](https://github.com/prusa3d/Slic3r/files/1295010/InputModel.zip)
* _Actual Results_
All internal size decreased by the width of the extrusion.
Side of the inner square was 9.6 mm. the Difference with the original size of 0.4 mm.
The inner diameter was 9.6 (radius 4.8 mm). The difference with the original 0.4 mm (10 mm diameter or a radius of 5 mm).
The external dimensions were not changed.
#### STL/Config (.ZIP) where problem occurs
[config.zip](https://github.com/prusa3d/Slic3r/files/1295043/config.zip)
[InputModelGCode.zip](https://github.com/prusa3d/Slic3r/files/1295046/InputModelGCode.zip)

bubnikv, 2017-09-12:
This is an evergreen, though it may be the first post at this issue tracker. To cut the story short, the Slic3r calculates the tool paths correctly, but the material properties make the extrusions around the hole to shrink. You may read the following topics to get the idea on what is going on and why it is difficult to compensate for:

https://github.com/alexrj/Slic3r/issues/1613
https://github.com/alexrj/Slic3r/issues/1588
https://github.com/alexrj/Slic3r/issues/348
https://github.com/alexrj/Slic3r/issues/56

http://reprap.org/wiki/ArcCompensation
http://forums.reprap.org/read.php?1,203510,203553
http://forums.reprap.org/read.php?1,45950

KostinPavel, 2017-09-12:
> alexrj#1613
> alexrj#1588
> alexrj#348
> alexrj#56

I know all the debates on this topic. But the problem is the place to be. It has been discussed in many forums! Not one free program layering does not provide a complete solution for this problem. slic3r has the compensation function of the dimensions in the XY plane. This feature compresses the size as on the outer and inner perimeters. A partial solution would be to split this compensation to separate compensation for the inner and outer perimeters. It would suit everyone: those who need this feature, and those who do not need this feature! Otherwise the idea with the promotion code Slic3r for me is not necessary. I think you can make a vote on this issue.

Hugoz12, 2017-09-12:
I like the idea of two separate values. 
The reason I don't use xy compensation is exactly that I don't want to have wrong outer perimeters for the correct inner perimeters (thus I just design everything with bigger holes etc.)

KostinPavel, 2017-09-12:
Two separate values are needed to meet the needs of all. It will not cure the real error of the inner perimeter. So for example when printing the perimeters of different widths, the error will manifest itself again. Therefore, it is more correct to solve the problem at the level of the perimeters of the layer.

KostinPavel, 2017-09-12:
Writing functions that separately adjust the dimensions of the inner and outer perimeters, makes it easy to solve problems at different extrusion width for perimeters in different layers in the future.

bubnikv, 2017-10-03:
The correct solution is to calibrate the hole shrinkage per filament, inner diameter, extrusion temperature etc. Without that, we are only providing a band aid. I may provide a separate hole expansion parameter, but as of now you will be better off to account for the hole shrinkage in your model. Sorry, but it is the current state of the art so long as nobody provided the aforementioned table of hole shrinkage.

What we really need in the first place is someone with a lot of time to print many holes with many filaments, temperatures and speed settings, and to derive some rules from that. Do you volunteer?

KostinPavel, 2017-10-03:
It's not the shrinkage! It is not in the holes!
Slicer does not work properly with the internal dimensions of the perimeters! Look carefully figure. Perhaps the problem will become clearer.
[image]

bubnikv, 2017-10-03:
> Slicer does not work properly with the internal dimensions of the
perimeters!

You are wrong. Slic3r plans the paths correctly.

KostinPavel, 2017-10-03:
For external perimeters all true, but for internal no. checked out a few real practices. I'm not the only one. many are faced with this problem. I'm not going to prove anything. Take my file and print. compare the result and the model. and you realize that you need a separate offset internal boundaries. everything else flood

bubnikv, 2017-10-03:
> I'm not going to prove anything. 

Hmm.

> I'm not the only one. many are faced with this problem. 

Any references?

KostinPavel, 2017-10-03:
I believe that the above information is enough to isolate and solve the problem.

fiveangle, 2017-10-04:
@KostinPavel - show us the STL + GCODE that proves the paths match your drawings :)

KostinPavel, 2017-10-05:
Gcode is good. the resulting internal size is bad. The resulting external size is good. The required method for the separate correction of the internal and external dimensions.See above. The cause of the phenomenon is not important. It is important that the phenomenon exists.

bubnikv, 2017-10-05:
> Gcode is good.

You see?

> The cause of the phenomenon is not important. It is important that the
phenomenon exists.

Do you have an advice for choosing the internal expansion parameter? If you
don't understand the phenomenon, you cannot reliably provide a formula for
the estimate of the required expansion parameter. If you don't understand
the phenomena, you have to find out the formula experimentally, which is
very time consuming and nobody did that yet. Do you volunteer?

KostinPavel, 2017-10-05:
while there is no formula, we need a mechanism of compensation with manual entering of values. when we have formula, we will use the formula. please make a manual payment the size of the interior perimeter. we write so many unnecessary words, and we simply need to make. and everyone will be happy.

KostinPavel, 2017-10-05:
if you are not able to implement, just say so. then I won't waste my time and yours.

macdarren, 2017-10-05:
A comment from the gallery...this has been a long running issue with slicers...for instance I am not sure why but Cura tends to give me better inside diameter matching than slicer but then slicer is better with other things.

As mentioned without understanding of what is happening it will be hard to create a true fix...but how about a work around...If inner dimensions can be detected by the slicer then maybe we could have 'correction factor'.  This would have to be determined by experimentation by each user...but could then probably be used for most similar prints.  

I am thinking something maybe that is multiples of perimeter width...maybe a way to skip 1 or more inner most perimeters or better just push them out in perimeter width steps removing them if they start to overlap the outer perimeters.

KostinPavel, 2017-10-05:
who stallcups with this problem. report it here. maybe only I have this problem? Maybe the topic is not relevant?

comps, 2018-02-16:
> The correct solution is to calibrate the hole shrinkage per filament, inner diameter, extrusion temperature etc.

I was wondering why the XY compensation is under "print settings" and not "filament settings", even though the shrinkage will mostly depend on the material (and its extruded temperature). Would it make sense to move the setting under "filament settings" (so that it is ie. applied to ABS and not PLA)?

nicksears, 2018-02-20:
This is still an issue and is not isolated to the type of material like you're thinking (vis-a-vi moving to filament settings). The post that mentioned inner diameters are an issue with Slic3r but not Cura sums it up perfectly in my opinion. This is a toolpath generation issue and I think it has to do with the fact that when extruding the elliptic (stadium) torus, the volume of the inner and outer half are the same, but this means the inner half extends further inward than expected. This would mean it can be mathematically compensated for...

Inner diameters are almost always too small in slic3r, but outer are usually fine. This can be exacerbated by mechanical issues, but even after addressing all of them here (http://manual.slic3r.org/troubleshooting/dimension-errors) and more, every printer will still have issues. The only workarounds I've heard of is deliberately underextruding external perimeters and polyholes, which are creative, but not the elegant solution we need.

XY compensation will fix inner diameters, but make outer diameters incorrect. I will hate to change the way I design just as much as everyone (and keep it in mind for modifying old designs where I've added significant diameter compensation) but this STILL needs to be fixed.

revoRay, 2018-09-19:
supermerill's addition of inner hole specific xy adjustment works really well. I hope it makes its way into the main branch.

RayVolts, 2018-09-19:
supermerill's addition of inner hole specific xy adjustment works really well. I hope it makes its way into the main branch.

KostinPavel, 2018-09-19:
The effect is all observed. But the solution is not implemented. I have to use other tools.

deathguppie, 2019-10-12:
I still don't understand why this is an argument.  A simple web search shows tons of people have this issue.  I have to turn my flow rate down until the extruded filament no longer touches its neighbors to get even close and then the outside is way too small.

The answer is to create a value to allow for inner hole size compensation so that the adjustment can be made in the slicer.  This doesn't affect people that don't have this problem and solves it for people that do.

Like I said, I seriously don't see why anyone would want to argue against the solution except that maybe some people just like to argue.

yodor, 2019-10-29:
Having the same issues as all above, I was testing @supermerill fork for 'hole compensation' the past few days. It works very nice for me. 
supermerill/Slic3r@dd20123

It seems the fork is adding also some other stuff but I'm interested only about 'hole compensation' I've prepared a small patch related only to this function. Applies to tag version_2.1.0 

[hole_compensation-version_2.1.0.zip](https://github.com/prusa3d/PrusaSlicer/files/3784909/hole_compensation-version_2.1.0.zip)

KostinPavel, 2020-01-16:
How can I get the final build for WINDOWS with this patch in mind? I want to test how this works in practice.

> Имея те же проблемы, что и все выше, я тестировал вилку @supermerill для «компенсации дырок» в последние несколько дней. Это работает очень хорошо для меня.
> [супермерилл / Slic3r @ dd20123](https://github.com/supermerill/Slic3r/commit/dd20123)
> 
> Кажется, что форк добавляет еще кое-что, но меня интересует только «компенсация дырок». Я подготовил небольшой патч, связанный только с этой функцией. Применяется к тегу version_2.1.0
> 
> [hole_compensation-version_2.1.0.zip](https://github.com/prusa3d/PrusaSlicer/files/3784909/hole_compensation-version_2.1.0.zip)

How can I get the final build for WINDOWS with this patch in mind? I want to test how this works in practice.

KostinPavel, 2020-01-16:
> Having the same issues as all above, I was testing @supermerill fork for 'hole compensation' the past few days. It works very nice for me. [...]

How can I get the final build for WINDOWS with this patch in mind? I want to test how this works in practice.

neophyl, 2020-01-16:
I don't know about the patch but you can get Supermerills fork here https://github.com/supermerill/Slic3r  its currently built on the PrusaSlicer 2.2.0 alpha2 so has almoist everything in (except I think the adaptive layer height stuff which looks like its missing).

KostinPavel, 2020-01-16:
Thanks for the hint. Downloaded here: https://github.com/prusa3d/PrusaSlicer/releases/download/version_2.2.0-alpha2/PrusaSlicer-2.2.0-alpha2+win64-201912221340.zip . The mechanism works. Half way done. It remains to add a similar correction for external perimeters.

supermerill, 2020-01-16:
> It remains to add a similar correction for external perimeters

If you want to change all perimeters, just use XY compensation.

KostinPavel, 2020-01-17:
> If you want a different settings for both, why not using a combination (-1 for external and -2 for internal -> -1 in XY compensation and -1 in "inner perimeter compensation") ?

You're right. This path can be chosen by me to get the desired result. So this question is related to usability. The use of the solution must be transparent to the user. This implementation is sufficient for understanding convenience.

6ffm70, 2020-07-31:
Interesting topic! I'm with @bubnikv - I think the printer is behaving correctly, but the material shrinks in either direction.

We've created a reference part (20mm³, 1,2,3,4mm wall thicknesses), that we print for every new material:
[image]

That then gets measured (digital caliper) outside and inside, and the compensation values calculated (not very scientific, but a viable practical approach giving good results):

[image]

Findings with the few samples:

- just one of 120 dim/s has the mean value
- hole dimensions tend to have a bigger deviation that outer
- material always _grows_ into the hole
- outer dims mostly smaller, but sometimes bigger
- only few filament types show similar behaviour

There are indeed unavoidable measuring tolerances, and environmental conditions that influence the result of these few samples. Also, we've printed with the nozzle that was already mounted (column E), and with different printers (column F). Slicer and parameters are always the same though.

While I started brainstorming on the ideal solution here #3687, I've added a possible solution for the outer dimensions shrinking compensation here #3059. The internal dimension compensation as discussed in this very issue is a whole different animal though, as it doesn't make sense to apply the same (scale) correction to outer and hole dimensions.

A possible (+/-) shrinking compensation stored in the `Filament Settings` would consider not only outer and hole dimensions, but also wall thicknesses, and maybe even infill/perimeter parameters.

nicksears, 2020-08-20:
Thanks @6ffm70 ! I'll just point out that I don't think this directly addresses the inner diameter of holes, but it does allow us to separate two issues.

This study seems to mostly show the effects of shrinkage, but based on your results, it seems to indicate that there's not a major contribution from shrinkage, so we don't need to worry about it (maybe a little for ABS).  

For hole diameters, we're worrying about the "corners". In a circle, there is one continuous corner. With polyholes, we seem to have developed a mathematical formula for the minimum number of corners to compensate for the filleted corner and NOT constrict the diameter.

We don't need any more proof than polyholes to tell use that we need software ID compensation. I have this on my flashforge knock-off, I don't know why it's so hard...

fbujold, 2020-10-29:
I have no clue who is right or wrong but there is a problem with the internal dimensions. I print a lot of aftermarket "knobs" and the inner hole is always wrongly dimension. If you look at the attached file the selector shaft is 6.7 mm. i had to draw it at 8 mm for it to fit properly.
[dryer button.zip](https://github.com/prusa3d/PrusaSlicer/files/5459241/dryer.button.zip)

nicksears, 2020-10-30:
see the attached example (my knock-off FFCP has this functionality) 
Notice that I have to shrink IDs because I still build in ~0.5mm ID expansion for my other printers [image]

KostinPavel, 2020-11-07:
Summing up the interim result. Internal dimensions will depend on many reasons. If we exclude the reasons related to the calibration of the material flow and the quality of the material itself, we get the following:
1. the dependence of the linear dimensions of the material on temperature. The main compensation method is scaling based on the print temperature and the expansion coefficient of the material. This may be a characteristic of the material. For many materials, the dependency is very weak.
2. method for constructing a trajectory for curved perimeters. The main compensation method is the construction of a piecewise linear trajectory for approximating the external and internal perimeters in various ways, taking into account the perimeter width. The question is probably similar to an inscribed and described circle.
3. the elasticity of the already extruded material when printing perimeters. Causes the contour to be rounded inside the printed object. This depends on the angle of change in the direction of the trajectory, and the width of the extrusion, and the temperature gradient along the length of the already printed perimeter, and the viscosity of the material, and its dependence on temperature, and so on. It is most pronounced at small sizes. The compensation method has not yet been determined.
4. other reasons. The compensation method was not defined. For example, you need to increase the wall thickness of thin-walled items.
How are these problems solved in slic3r?
1. scale the object before you can break it down into layers. This is very inconvenient, as it requires manual scaling. Rarely used, but universal.
2. There is no solution. This can be a selectable setting.
3. There is no solution.
4. There is no solution.

boelle, 2021-04-13:
i did my esteps, flow calibration and still all my X and Y is off by 0,5625%

i know things shrink and i would have used the XY size compensation if it was not that this is locked to mm

what is the most correct way to compensate?

ctadlock, 2021-05-13:
Is this issue ever going to be addressed? It seems that having separate X/Y compensation for internal and external perimeters would be a straightforward feature that would go along way to resolving this issue.

Also X/Y compensation should allow %s, not just absolute numbers.

boelle, 2021-05-13:
it would be nice to have it adressed, but given that it was raised in september 2017 i dont think it will be

a shame thou, and yes it needs to be in % to as nothing with 3d printing is the same absolute, but usual numbers in % are :-D

nicksears, 2021-05-13:
I disagree that it needs to be in %, flashprint gives internal and external compensation and it works like a dream. Seems like it should be trivial...
[image]

fiveangle, 2021-06-01:
> Seems like it should be trivial...

Go for it then ;)

nicksears, 2021-06-01:
Unfortunately, my solution has just been to use other software ¯\_(ツ)_/¯

bubnikv, 2021-06-02:
To make such a compensation algorithm is quite tricky for example on the following object, showing two successive slices. On the left the red contour shows a hole. On the right the hole disappears, everything is a single contour.

[image]

Our Superfriend implemented the following algorithm in his Superfork, adding three new parameters to the existing one:

xy_size_compensation
xy_inner_size_compensation
hole_size_compensation
hole_size_threshold

If a contour is identified to be a hole (negative area, only concave corners):
	1) If its area is smaller than 1x hole_size_threshold, apply the following compensation: xy_inner_size_compensation + hole_size_compensation
    2) If its area is larger than 4x hole_size_threshold, apply the following compensation: xy_inner_size_compensation
    3) If its area is in between, apply a compensation value linearly compensated by the area. Here it makes more sense to interpolate by the square root of the ratio, but I did not try to print with it.

It somehow makes sense to me the way it has been implemented and I suppose this implementation went much further than the original suggestions after rounds of testing. The compensation algorithm does not really detect a circular hole, it applies the same rule to rectangles, thin long trapezoids etc. IMHO there should be at least some rule on eccentricity. Do we want to add another parameter to the UI layer? How many parameters do we need to describe the algorithm? Will the customer understand it? What is the risk that the algorithm compensates some holes incorrectly? What is the risk that the compensation algorithm will create discontinuities on the object surface?

boelle, 2021-06-02:
> I disagree that it needs to be in %, flashprint gives internal and external compensation and it works like a dream. Seems like it should be trivial...
> [image]

i did not mean that it HAS TO BE, but as an option for those that want to use % instead of a set number

bubnikv, 2021-06-04:
Frankly I see a utility in such functions, but implemented in this simplified form they are a double edged sword. The vast majority of our users will never use it and if we make such feature publicly available thus "supported", we would face issues as discontinuities or unexpected behavior. I am considering to add an experimental, not supported mode, where these options would pop up, but don't call our support if it does not work for you.

boelle, 2021-06-04:
> but don't call our support if it does not work for you.

I dont have a prusa machine so will not call them

fiveangle, 2021-06-08:
> I am considering to add an experimental, not supported mode

That seems a perfect compromise if you can cleanly identify the feature as experimental/unsupported within the UI. It should obviously be done in such a way that the same method could be used for other potential experimental/unsupported features in the future.  This would allow PS to maintain relevance against other slicers that are pushing bleeding edge features, while not abandoning the solid foundation that the more general users expect from PS.  Nice thinking @bubnikv 👍💯

padand, 2021-06-11:
@bubnikv You do raise some good usability points and perhaps I can answer some of your questions, as I've been dealing with this problem for years. I mostly print functional parts that I 3D model myself and I've been modifying all my models manually to compensate for these differences. After all this time, I found out there are three types of areas I need to resize, which should be easy to understand from this diagram:

[image]

So, for example:

- If I print I clamp, I make it 0.3 larger to fit around an object
- If I print a shaft, I make it 0.2 smaller to fit inside a given hole
- If I print a hole, I make it 0.4 larger to fit a given shaft

Regarding the shape of the hole, you are on to something. I can tell you that the key factor is symmetry, not shape. Square, circle and hexagon behave the same. After all, from a 3D modelling perspective they are all "circles" with a different number of segments. Stretched holes tend to behave more like concave perimeters along the stretch and holes at the ends. Still, the difference between the two are very small so I don't think it's wise to take the algorithm to such an extent. I think it would actually have a negative impact on usability. This, like you said, is not for the vast majority of users. It's an "expert" level feature. And as a (self proclaimed) expert, I can tell you I am not looking for a complex compensation mechanism that would cover 99% of all the scenarios. I just want a feature that would spare me from the tedious task of resizing (or worse, grinding) in maybe 80-90% of the cases. I am happy to deal with the exceptions manually, as long as the algorithm is simple and intuitive enough so that I know exactly what it's doing and I can take that into account. I think having two params like perimeter compensation and holes compensation would be simple enough in this regard and it's what others are doing. Being able to differentiate between convex and concave perimeters would be gold! I wouldn't even go as far as having a threshold for the holes, as it's already becoming non-intuitive.

You also mentioned something about discontinuities on an object's surface. There were a very rare number of occasions where I experienced that, but it was due to some perimeters close to a hole becoming too thin. Not sure if this is what you had in mind. Still, bear in mind that this feature would be used mostly for functional parts, where a good fit is more important than looks. If someone wants to print some artsy stuff, they can easily turn this off or never have to deal with it in the first place. Keeping it an expert level feature should help with that.

All in all, I think a simple implementation like this would not only be better from a usability standpoint, but also less likely to raise issues. Thanks for taking the time to read this!

BrienAllison, 2021-06-19:
I was using PS today and was surprised there was no hole contour compensation which I find very useful. I am printing one of my [iris boxes](https://www.myminifactory.com/object/3d-print-12-leaf-aperture-iris-box-87618). The doors of the box use small dovetail joints to slide together. In order to get a nice round shape when it is closed, the sliding joint needs to stop at a very specific point, otherwise it doesn't close neatly and looks uneven. This is impacted strongly by the outer contours since:

1. The contour errors on both parts add up.
2. The meeting surface is at 60 degrees to the dovetail motion, which has a wedging effect that doubles the impact of contours on sliding joint misalignment.

Both of these properties together means that outer contour errors impact the closed postion of the sliding joint by a factor of 4 e.g. a contour error of 0.1mm produces a positional error of 0.4mm which doesn't sound like a lot but is very noticable.
This effectively means the outer contours need to be finely tuned in their own right for specific printing parameters such as temperature and layer height.

What complicates the issue is that the same part has holes for screws which also need to be appropriately sized, so a separate value for hole compensation is ideal.

Although with the parts I printed today without contour compensation in PS worked fine, since I have designed and tested the part with standard settings in mind. It doesn't work for all situations however e.g. lower layer heights etc. Particularly when the design is scaled up 150% to use larger screws, the contour compensation needs to be tuned again, including the holes which are generally far too tight at first.

Of course one solution is to redesign the part for different sizes, which is fine for me since I designed it, but for other people it is not possible, and producing many variations of the same part complicates things for people downloading it. Currently I am recommending Idea Maker as a slicer since it is one of the free options which can do hole and contour compensation, but I am currently transitioning to PS but this issue is preventing me converting entirely. 

> I think having two params like perimeter compensation and holes compensation would be simple enough in this regard and it's what others are doing. Being able to differentiate between convex and concave perimeters would be gold!
 Exactly my thoughts. This doesn't need to be a perfect implementation which works with any shape all the time, just having inner and outer contour options for starters is handy. 

The convex and concave perimeters identification would meet my needs 99% of the time. I have experienced this open/closed perimeter transition issue already in Idea Maker, but most of the time the trade off is worth it. 

One approach would be to use a cylinder modifier in the hole, and have an option to adjust the outer contour compensation accordingly so that the contours match, even with open and closed contours. IM also has shape modifiers, but contour compensation is not adjustable. This would also only really work with simple straight holes though, and I don't know how hard that would be to implement. Maybe it could be possible to make paint on surface modifiers, like paint on supports, but they adjust properties of the contours? That would be amazing. Idea Maker has added the surface texturing recently, and that works in a somewhat similar sense, using colours to adjust contours.

jeedee23, 2022-01-05:
As I read all of the above, this issue is still hanging? 
I don't find the **inner XY compensation** or the **outer XY compensation** just the **XY compensation**

so the print can just be completely shrunk or completely expanded on XY, right? That is a big problem since we printed a bunch of nylon gears, with an inner polygon (or axle or whatever hole), and all of them are wrong. 
[image]
Where you see 16.7 mm on the image, has a difference of .7 mm, so the XY **INNER** should be expanded **outwards** by .35 mm. 
The outer diameter is correct by 0,05 mm which is normal for 3D printing.

the quote here above is correct:

`I think having two params like perimeter compensation and holes compensation would be simple enough in this regard and it's what others are doing. Being able to differentiate between convex and concave perimeters would be gold!`

Please please resolve this! Too long, too long.

regards
Johan

adamfilip, 2022-07-16:
Why is this closed, Hole Compensation issue has not been resolved. and Issue #7719 is also closed and says to reference this thread.  #515

jiridanek, 2022-07-16:
@adamfilip that `Closed` above refers to the X and Y compensation issue referencing this. The issue we are now commenting is open, as indicated by the Open chit on top of the page (I use colorblind theme, so the colors would seem weird to you)

[image]

leon-titan, 2022-07-16:
Need this feature too, could you please add this feature soon?

danoveb, 2022-07-20:
I have no idea how hard it would be to implement, but my wish is the current XY compensation found in Superslicer, but with multiple parameters where you set a compensation value for different radius and then interpolate between this values. I think 4-5 values would be enough. I changed to a 0.8 Nozzle a year ago, and haven't went back to 0.4 since, but I need to adjust the holes for this already  in the design. For me, it's not about compensation for the shrinkage, but only for the "corner cutting". I understand that this has been an unsolved issue for a long time (https://reprap.org/wiki/ArcCompensation), but I think my suggest may be a "good enough" solution. Today I have done a Excel sheet that do the same thing, but I need to change the radius during design time. The backside of this is if I want to change to a 0.4 Nozzle, I need to change all the hole radius...

jeedee23, 2022-07-20:
Isn't this what you need, since you are talking about holes, the reprap link talks about curves.
[image]

danoveb, 2022-07-21:
Thank you for pointing that out. Not exactly that I had on my mind, but it might be good enough. Will test that out. Hopefully, SS will catch up with PS so we get variable line width in SS too. About the ArcCompensation, holes are curves right? curves with a low radius have the same issue as the holes...?

danoveb, 2022-07-21:
Well it's actually Slic3r -> PrusaSlicer -> Superslicer. So Superslicer is a fork of PrusaSlicer which is a fork of Slic3r... Anyway, because I already had the "correct" compensation values in a Excel sheet, I made a formula for the SS (Why is the threshold in mm2?) function and reversed engineered the settings and did a test print and it actually worked quite well, but not perfect. For my 0.8 Nozzle this settings was spot on for 3-6 mm holes, the bigger was constantly + 0.2 mm too big. XY Compensation Inner mm = 0.2, XY Compensation mm = 0.2, Threshold mm2 = 45. Tried alter the settings, but this gave the best result.

KarlGerhardt, 2022-08-25:
Just want to cast my vote for the addition of this feature, for whatever it's worth. It seems the position of the devs is 'adjust the model'. This is an onerous task when you have many features. Especially if all you have is an .stl. Most slicers have this feature now.

apphead, 2023-01-24:
'adjust the model' is not an option if it is a complex stl file from thingiverse etc. This is a must have high priority feature for my applications. Please add it asap! Thank you Prusa!

elephantwalker50, 2023-01-26:
I opened another issue on this... 

modeling is much too complicated to fix all internal dimensions. I'm already using wall thickness in models that are reproduced well in the slicer. however, the issue with internal dimensions of holes breaks the models. 

i am actively searching for software that can solve this problem.

please no arguments. just fix it.

[28 more replies]

closed 2026-09-03 by KostinPavel