prusa3d/PrusaSlicer #519: Smarter infill around holes? (Suggestion)

joeynuggetz, 2017-09-13:
I've always been curious why slicers seem to lay down infill around screw holes in a way that promotes surface scarring and longer travel moves. Is there a problem for the infill to take the shortest path to the next area requiring infill instead of infilling around the screw hole and then having to travel back near the end of the layer to come back and fill the void?

I've always been curious why slicers seem to lay down infill around screw holes in a way that promotes surface scarring and longer travel moves. Is there a problem for the infill to take the shortest path to the next area requiring infill instead of infilling around the screw hole and then having to travel back near the end of the layer to come back and fill the void?

[image]

bubnikv, 2017-09-14:
Isn't your issue a duplicate of https://github.com/prusa3d/Slic3r/issues/471
? I believe the issue https://github.com/prusa3d/Slic3r/issues/471 gives
you the answer to your question.

bubnikv, 2017-11-28:
Duplicate of #471, closing this one.

closed 2017-11-28 by bubnikv