prusa3d/PrusaSlicer #520: Allow gaps smaller than 100 microns

DrLex0, 2017-09-15:
### Version
1.36.2

### Operating system type + version
Any

### Behavior
While trying to slice https://www.thingiverse.com/thing:1356547 (the _Pyramid5_ model) as per the instructions, I noticed vase mode simply does not work for this model. It isn't printed in vase mode because what Slic3r actually slices, is not the same as what is in the actual 3D model, some filtering operation is applied before slicing (and yes, I set the ‘resolution’ in the advanced settings to 0).

If you look at that model carefully, you'll see it has many tiny gaps to ensure it can be printed in spiral vase mode.
[image]

Cura does not ignore those gaps, which makes vase mode work there. Slic3r however fills all those gaps, which causes many parts of the model to become solid. The gaps are visible in the 3D view, but not in the gray outline of the layers view, which I guess represents the shape that is actually sliced.

I did some tests with a simpler model: [GapTest.stl.zip](https://github.com/prusa3d/Slic3r/files/1307771/GapTest.stl.zip)
This has 7 gaps, the smallest of which is 90 microns wide, while the second smallest is 100 microns. If this is _Slic3d_, that 90 micron gap just disappears no matter which settings I use:
[image]

In most cases this is probably desirable, but in some cases it would be nice to have an advanced option to lower the minimum gap size. Not just for this model; I already bumped into the same problem with some of my own models where I wanted to trick Slic3r into printing internal structures in one continuous line together with external perimeters.

bubnikv, 2017-10-03:
ok, I have found it. There is a closing operation performed over each slice with the default ball radius of 0.0499mm. That means, gaps smaller than that will be closed.

It is in TriangleMesh.cpp:1210. If you replace the number with 0.0001, then the fractal pyramid will slice correctly.

Now I wonder what I shall do with that. If I just lower the number, I may expect new issues with some models. If I just add another parameter, it will make the Slic3r yet a bigger mess than it is right now. Any idea?

Frankly I am not sure why the closing operation is there in the first place, so I am afraid to touch that as thousands of people use it.

DrLex0, 2017-10-03:
I guess this is some protection against poorly made models with unintentional tiny gaps that would cause strange tool paths or weaknesses.
I also wouldn't just omit or change this, but still it would be nice to make it adjustable, not just for this use case. When printing with smaller nozzles, 0.1mm gaps may become significant, and detail would be lost if they would be filled. Of course this setting should be well buried in an advanced panel.

bubnikv, 2017-10-30:
Referring a modified fractal pyramid :-)
https://www.thingiverse.com/thing:2613568

DrLex0, 2017-10-30:
Yes, originally I planned to make this kind of OpenSCAD model from scratch, but someone else beat me to it. I only improved upon the model :-)

kortschak, 2018-07-04:
@bubnikv if the threshold is lowered for the case when the model is to be sliced in vase mode you should be able to depend on the assertion by the user that model is a single manifold surface. Then you can check after slicing that no travels were made and otherwise report an error.

Goury, 2019-01-02:
Seems to be fixed in 1.42

bubnikv, 2019-03-01:
By the way, I have introduced a new parameter "slice_closing_radius", which is set to 0.049 by default, leading to the same results as in 1.41.3. If you need to maintain the thin gaps, starting with Slic3r-1.42.0-alpha7 you will need to lower the new "slice_closing_radius" value.

Implemented with e3cba0a65a691b77be87f8440c91ec5f4916c3b8

This parameter and the return to the 1.41.3 behavior by default was made as we have discovered some models with cracks, which were handled correctly by 1.41.3, but which were full of spurious perimeters in 1.42.0-alpha5. We find it safer to revert to the 1.41.3 behavior while giving you the parameter to tinker with.

closed 2019-03-01 by bubnikv