prusa3d/PrusaSlicer #521: Vase mode Issue

Serray, 2017-09-16:
### Version

Version 1.36.2-prusa3d-stable-win64

### Operating system type + version
Windows 10 64bit

### Behavior

If I want to print an object in Vase mode, whos model is not solid (like a cube) but has an inner and an outer perimeter (like a Glass) it wont be printable.
After the solid bottom layers, it will print one perimeter perfectly.
But after some layers, it starts to print an inner perimeter into the air,
causing a filing print and seams on both the inner and outer perimeter, because it has to switch between them.

bubnikv, 2017-10-03:
Would you please provide an STL?

The vase mode is only intended for closed models, where the vase mode extracts a single continuous outer contour only. It will not work on hollowed models like a glass. It will work, if you close the glass with a lid and fill in to make it solid in your solid modeler.

Please check the models on thingiverse or elsewhere marked to be printable in a vase mode.

bubnikv, 2018-02-12:
No response, closing.

closed 2018-02-12 by bubnikv