prusa3d/PrusaSlicer #525: Program crashes when trying to "Quick slice and save" on file on network storage device

oleost, 2017-09-20:
### Version
Version 1.36.2-prusa3d-stable-win64

### Operating system type + version
Windows 10

### Behavior
* The program force closes when trying to do "Quick slice and save" when stl is on a network storage device.

* Use "Quick slice and save" on a file on example an windows share.
* It will save new gcode
* Slic3r crashes after you press "Save"

[image]
[image]
[image]
[image]

#### STL/Config (.ZIP) where problem occurs
[22a-fender-right.zip](https://github.com/prusa3d/Slic3r/files/1316846/22a-fender-right.zip)
[config.zip](https://github.com/prusa3d/Slic3r/files/1316853/config.zip)

bubnikv, 2017-09-20:
Is the crash specific to your config & stl, or is it specific to the
network storage?

I tried to reproduce using some small STL and a network storage running a
Linux Samba software, but I did not get any crash.

You may also try to run the slic3r-console.exe with the parameter --gui
from a command line to get some text output. The text output may indicate a
line in the Perl source code, where Slic3r crashed.

oleost, 2017-09-25:
Sorry for late respond, tried using slic3r-console.exe but I didnt get any error message from it either.

Will have to try and troubleshoot some more if your unable to reproduce it.

bubnikv, 2017-10-03:
Please try to find a way for me to reproduce the issue, otherwise I will have to close it. Thank you.

oleost, 2017-10-03:
I'm unable to reproduce it now my self, even through I could reproduce it easily last time.

Anyway, I`m closing this issue then. If it happens again for me I will try and get some more details and create a new issue.

Appologize for wasting your time.

closed 2017-10-03 by oleost

bubnikv, 2017-10-03:
You are certainly not wasting my time. I know very well how difficult are
some issues to reproduce.
Thanks for your effort.