prusa3d/PrusaSlicer #526: Freeze when

jojo313, 2017-09-22:
### Version
_1.36.2 64bit_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_

### Behavior
* _Problem: Slic3r freezes and uses 100% CPU and Disk when choosing Archimedian Chords. (on a i5 4590k and SSD)_
* _Choose Archimedian Chords Fill_
  * _If this is a command-line slicing issue, include the options used_
* _Slices with Archimedian Chords fill_
* _Freezes and uses 100% CPU and Disk_
  * _http://prntscr.com/gogwhf_

_No_

#### STL/Config (.ZIP) where problem occurs
_[config.zip](https://github.com/prusa3d/Slic3r/files/1325564/config.zip)/all STL_

bubnikv, 2017-10-03:
I am not able to reproduce the issue. Would you please provide the STL? By the way, you don't have the achimedian chords enabled in your config, but the concentric infill.

jojo313, 2017-10-03:
Yes. Because of I select it it crashes... Also happens on all STL I tested.

Sent from ProtonMail mobile

-------- Original Message --------

bubnikv, 2017-10-03:
It does not happen on my computer.

do you have the background processing enabled? Does slic3r reslice automatically at each parameter change, or do you have to click the "Slice" button manually?

If you still have the issue, would you please try to describe the sequence of steps in detail, so I can try to reproduce your crash?

jojo313, 2017-10-03:
I'll try it on a different system. And yeah background and automatic slicing should be enabled (by default?).

Sent from ProtonMail mobile

-------- Original Message --------

bubnikv, 2017-10-03:
> prusa3d/Slic3r/issues/526#issuecomment-333919887), or [mute the thread](
> AettI9IEXVkbsypn9LZVCVM8cf1ipE6qks5som-9gaJpZM4PhHwC).

We have the background processing disabled by default.

jojo313, 2017-10-03:
That's weird. Probably came from my old slicer installation. Just realize I should have clean installed that... Fuck me...
Sry for wasting your time will reply when I've tried.

Sent from ProtonMail mobile

bubnikv, 2017-10-21:
probably an installation issue, closing.

closed 2017-10-21 by bubnikv