prusa3d/PrusaSlicer #527: Compilation failed: bareword "STEP_WIPE_TOWER" not allowed

rockstorm101, 2017-09-24:
### Version
Custom 1.37.0 which implements commits https://github.com/prusa3d/Slic3r/commit/888a904c9b484289318737ec583ac0794dfa84f7 and https://github.com/prusa3d/Slic3r/commit/6d4ec5c989851e442d58f896d4b0324129e7514a.

### Operating system type + version
Debian GNU/Linux unstable (sid)

### Behavior
The command line seems to work but when I try to initialize the GUI I get this error:

```
$ slic3r-prusa --gui
Bareword "STEP_WIPE_TOWER" not allowed while "strict subs" in use at /usr/lib/slic3r-prusa/Slic3r/GUI/3DScene.pm line 1967.
Compilation failed in require at /usr/share/perl/5.26/base.pm line 136.
	...propagated at /usr/share/perl/5.26/base.pm line 158.
BEGIN failed--compilation aborted at /usr/lib/slic3r-prusa/Slic3r/GUI/Plater/3D.pm line 11.
Compilation failed in require at /usr/lib/slic3r-prusa/Slic3r/GUI.pm line 22.
BEGIN failed--compilation aborted at /usr/lib/slic3r-prusa/Slic3r/GUI.pm line 22.
Compilation failed in require at (eval 85) line 1.
```
I'd be happy to provide any further info on the matter, just tell me what you need.

bubnikv, 2017-10-03:
I think you have the XS modules mixed somehow, so the Slic3r Prusa Edition is either loading the XS module of the mainline Slic3r (by Alessandro Ranelucci), or it is loading an XS module of an older Slic3r PE.

rockstorm101, 2017-10-03:
I am no longer able to reproduce this with version 1.37.1, so I'm closing the issue. It was most likely due to cherry picking commits. My apologies.

closed 2017-10-03 by rockstorm101

bubnikv, 2017-10-03:
no problem. thanks for heads up.