prusa3d/PrusaSlicer #529: Assertion failures while running tests in 1.37.1

hyperair, 2017-09-25:
### Version

1.37.1

### Operating system type + version

Debian unstable (also fails in Ubuntu zesty)

### Behavior

Tests are failing due to assertion failures.
```
        cd obj-x86_64-linux-gnu && make -j4 test ARGS\+=-j4
make[2]: Entering directory '/<<BUILDDIR>>/slic3r-prusa-1.37.1+dfsg2/obj-x86_64-linux-gnu'
Running tests...
/usr/bin/ctest --force-new-ctest-process -j4
Test project /<<BUILDDIR>>/slic3r-prusa-1.37.1+dfsg2/obj-x86_64-linux-gnu
    Start 1: xs
    Start 2: integration
1/2 Test #1: xs ...............................   Passed    2.41 sec
2/2 Test #2: integration ......................***Failed   31.66 sec
Attempting to create directory /<<BUILDDIR>>/slic3r-prusa-1.37.1+dfsg2/t/../local-lib
t/angles.t ..................... ok
t/avoid_crossing_perimeters.t .. ok
t/bridges.t .................... ok
t/clean_polylines.t ............ ok
t/clipper.t .................... ok
t/collinear.t .................. ok
t/combineinfill.t .............. ok
t/config.t ..................... ok
t/cooling.t .................... ok
t/custom_gcode.t ............... ok
t/dynamic.t .................... skipped: variable-width paths are currently disabled
perl: /<<BUILDDIR>>/slic3r-prusa-1.37.1+dfsg2/xs/src/libslic3r/GCode.cpp:850: void Slic3r::GCode::process_layer(FILE*, const Slic3r::Print&, const std::vector<Slic3r::GCode::LayerToPrint>&, const Slic3r::ToolOrdering::LayerTools&, size_t): Assertion `! layer_tools.extruders.empty()' failed.
t/fill.t .......................
Failed 4/19 subtests
        (less 1 skipped subtest: 14 okay)
t/flow.t ....................... ok
t/gaps.t ....................... ok
t/gcode.t ...................... ok
t/geometry.t ................... ok
t/layers.t ..................... ok
t/loops.t ...................... skipped: temporarily disabled
t/multi.t ...................... ok
perl: /<<BUILDDIR>>/slic3r-prusa-1.37.1+dfsg2/xs/src/libslic3r/GCode.cpp:850: void Slic3r::GCode::process_layer(FILE*, const Slic3r::Print&, const std::vector<Slic3r::GCode::LayerToPrint>&, const Slic3r::ToolOrdering::LayerTools&, size_t): Assertion `! layer_tools.extruders.empty()' failed.
t/perimeters.t .................
Failed 1/59 subtests
t/polyclip.t ................... ok
t/print.t ...................... ok
t/retraction.t ................. ok
t/shells.t ..................... ok
t/skirt_brim.t ................. ok
t/slice.t ...................... skipped: temporarily disabled
t/support.t .................... skipped: temporarily disabled
t/svg.t ........................ ok
t/thin.t ....................... ok
t/threads.t .................... ok

Test Summary Report
-------------------
t/fill.t                     (Wstat: 6 Tests: 15 Failed: 0)
  Non-zero wait status: 6
  Parse errors: Bad plan.  You planned 19 tests but ran 15.
t/perimeters.t               (Wstat: 6 Tests: 58 Failed: 0)
  Non-zero wait status: 6
  Parse errors: Bad plan.  You planned 59 tests but ran 58.
Files=30, Tests=382, 31 wallclock secs ( 0.10 usr  0.04 sys + 32.74 cusr  1.02 csys = 33.90 CPU)
Result: FAIL

50% tests passed, 1 tests failed out of 2

Total Test time (real) =  31.66 sec

The following tests FAILED:
          2 - integration (Failed)
Errors while running CTest
Makefile:97: recipe for target 'test' failed
```

bubnikv, 2017-09-25:
This is due to the new CMake based build system, which compiles in debug
mode by default. You should call cmake with the following parameter:

 -DCMAKE_BUILD_TYPE=Release

I know I should document it somewhere, but there is so much to do. As of
now, I just documented it for the Windows build, as on Linux the CMake
based build is fairly straight forward.

https://github.com/prusa3d/Slic3r/wiki/How-to-compile-Slic3r-Prusa-Edition-on-MS-Windows

Thanks for taking care of the Debian package.

hyperair, 2017-10-02:
This fixed it, thanks.

closed 2017-10-03 by bubnikv