prusa3d/PrusaSlicer #531: Ooze prevention not working correctly on single extruder print

Stopka, 2017-09-30:
### Version
1.36.2, built from sources using aur package

### Operating system type + version
Arch Linux

### Behavior
* I have dual extruder printer. I have enabled ooze prevention with temp variation -40 degrees C. When printing with both colors, it works as expected: Printer preheats both nozzles to the _{filament temp}-{ooze temp variation}_ and when starts printing, it selects extruder and heats up selected nozzzle to the _{filament temp}_. Problem is when I print with one color only. In this situation it preheats the default nozzle to _{filament temp}-{ooze temp variation}_ an then starts printing without heating up to the _{filament temp}_
* _Steps needed to reproduce the problem
  * Enable ooze prevention
  * Set Temperature variation to -40
  * Set filament temperature to  255
  * Print using only one (default) extruder
* _Expected Results_
  * It is printing with the nozzle heated up to temperature of 215
* _Actual Results_
  * It should preheat nozzle to 215 and then before printing heat up to 255 

_Is this a new feature request?_
It is gcode genering bug i think

#### STL/Config (.ZIP) where problem occurs
[STL+config.zip](https://github.com/prusa3d/Slic3r/files/1346310/STL.config.zip)

bubnikv, 2017-10-03:
By the way, when printing on a dual extruder machine with a single extruder only, how do you expect the other extruder to be heated?

Stopka, 2017-10-03:
Now the situation is that the not used extruder is not heated at all and I am ok with that behaviour.

bubnikv, 2017-11-28:
Now I am confused. You are saying, that the latest Slic3r Prusa Edition build (1.37.2) does not heat up the unused extruders? This happens if:

1) You have a custom start G-code script, where you set some extruder temperature manually.
2) You have the "single extruder multi material" checkbox checked.

In any other case Slic3r will currently generate the extruder preheat commands for all extruders.

Would you please expand on what you expect Slic3r do in regard to multiple extruders and temperatures?

Stopka, 2017-11-29:
I am saying the property 'enabled ooze prevention' together with property 'ooze prevention/temperature variation' causes the nozle not to heat up to full temperature when printing on one nozzle only..

closed 2026-09-29 by prusaslicer-bot[bot]