prusa3d/PrusaSlicer #532: admesh dependency breaks slic3r-prusa for all big-endian architectures

hyperair, 2017-09-30:
### Version

1.31 - 1.37

### Operating system type + version
Ubuntu

### Behavior

Slic3r fails to compile: https://buildd.debian.org/status/fetch.php?pkg=slic3r-prusa&arch=mips&ver=1.37.1%2Bdfsg2-1&stamp=1506540831&raw=0

```
/<<BUILDDIR>>/slic3r-prusa-1.37.1+dfsg2/xs/src/admesh/stl.h:32:2: error: #error "admesh works correctly on little endian machines only!"
 #error "admesh works correctly on little endian machines only!"
  ^~~~~
```
_Is this a new feature request?_

No

hyperair, 2017-09-30:
Actually I just noticed that there's an admesh package in Debian that builds on BE architectures. I'm going to patch slic3r-prusa to use that instead.

hyperair, 2017-10-01:
@bubnikv did you find any issues in admesh on big-endian architectures besides binary importing? I'm working on a fix for that now so we can drop the `#error`.

bubnikv, 2017-10-01:
> did you find any issues in admesh on big-endian architectures besides
binary importing? I'm working on a fix for that now so we can drop the
#error.

Frankly I don't remember anymore. The vast majority of today's platforms
are little endian, so I did not really care about the big endian machines.

I think the biggest issues with the original admesh lib was the handling of
the positive / negative zeros, which are considered equal when comparing
with a floating point unit, but they are different when casted to uint32.

Also there were some issues in regard to the ASCII import I believe, but I
don't remember the details.

bubnikv, 2017-10-03:
Big endian fixed with https://github.com/prusa3d/Slic3r/commit/a52a04550e2ee869712ac340bb4d459ba6be3058

closed 2017-10-03 by bubnikv