prusa3d/PrusaSlicer #533: MMU First Layer Temperature Issue

cosmith71, 2017-10-01:
### Version
1.37.1-prusa3d-win64 Multi-Material

### Operating system type + version
Windows 10 Pro

### Behavior

When using different extruder temperatures for extruder 4 and extruder 1, such as for soluble supports, the first layer prints at the extruder 4 temperature.

Extruder 4, set to 190 for PVA
Extruder 1, set to 205 for PLA

Extruder heats to 190, prints test strips, first layer of wipe tower, and raft at 190 regardless of extruder used.  Temperature remains at 190 when switching to extruder 1, which should be at 205.  Temperature appears correct when moving to layer 2.

#### STL/Config (.ZIP) where problem occurs
[robot_articulated.zip](https://github.com/prusa3d/Slic3r/files/1347074/robot_articulated.zip)

bubnikv, 2017-10-03:
I believe it is a new issue, which has been introduced with the new extruder priming areas at the front edge of the print bed. Am I correct?

I hope I have fixed that with
https://github.com/prusa3d/Slic3r/commit/e719e48a84af32cda1cd3af35bbe10813aa2a299

bubnikv, 2017-10-22:
It has been fixed with https://github.com/prusa3d/Slic3r/releases/tag/version_1.37.2

closed 2017-10-22 by bubnikv