prusa3d/PrusaSlicer #535: Rendering Fails on Multi-Extruder Slices with Wipe Tower

bardiir, 2017-10-04:
### Version
1.37.1-prusa3d

### Operating system type + version
Max OS 10.11.6

### Behavior
When I configure the attached cube with one part printed with Extruder 1 and one part printed with Extruder 3 my Rendering fails and I've no longer got a plate. The issue does not show up when there is no Wipe Tower.

#### STL/Config (.ZIP) where problem occurs
[issue.zip](https://github.com/prusa3d/Slic3r/files/1356781/issue.zip)
[image]

bubnikv, 2017-10-04:
Good we have an issue for it now :-)

I hope I have fixed it couple of hours ago.
https://github.com/prusa3d/Slic3r/commit/a191fbbec8d0794a23d0b1530f70b2bcf979f4ee

bardiir, 2017-10-04:
Hmm I really need to look into how to compile my own Slic3r version then I guess.

bubnikv, 2017-10-21:
Fixed with https://github.com/prusa3d/Slic3r/commit/a191fbbec8d0794a23d0b1530f70b2bcf979f4ee
It will be part of the upcoming 1.37.2 patch release.

closed 2017-10-21 by bubnikv