prusa3d/PrusaSlicer #536: Cooling Threshold Speed Inconsistancy

m-mccormick, 2017-10-05:
### Version
1.37.1

### Operating system type + version
Windows 7 x64

### Behavior
When printing layers with a small cross-sectional area and therefore low print time, external perimeters are not scaling down to be a slower speed with the rest of the layer due to "Cooling Threshold" settings under the Filament Settings tab. Decreases in speed due to decreases in layer print times worked flawlessly in 1.34.1, and I am running into this issue only after upgrading to 1.37.1. All other speeds seem to be scaling in proportion to layer print time as expected; Only external perimeters are fast, and appear to be going at the "Perimeters" speed judging qualitatively by sound and visual feedback.

#### STL/Config (.ZIP) where problem occurs
Attached
[ConfigAndStl.zip](https://github.com/prusa3d/Slic3r/files/1357677/ConfigAndStl.zip)

tcm0116, 2017-10-07:
I can second this issue. When the print speed is decreased due to Auto Cooling, the external perimeter speed is not reduced.

bubnikv, 2017-10-07:
See the discussion at the upstream Slic3r.
https://github.com/alexrj/Slic3r/issues/4070

The current behavior is as intended. We can now discuss, how it is supposed to work.

bubnikv, 2018-04-25:
We have found that the current cooling logic does indeed cause a lot of issues with our i3 MK3 printer, see  prusa3d/Prusa-Firmware#602 , and we are currently reworking the cooling logic.

bubnikv, 2018-08-02:
The cooling logic was reworked in May. Closing.

closed 2018-08-02 by bubnikv