prusa3d/PrusaSlicer #537: Slic3r crashes on start, if previously closed on second monitor

jnbr18, 2017-10-05:
### Version
1.37.1-prusa3d-win64

### Operating system type + version
Win10 Pro, 64bit
Details: [System Information.txt](https://github.com/prusa3d/Slic3r/files/1357655/System.Information.txt)

### Behavior
When I close Slic3r on my second monitor it crashes (stops working) when I start it again.
When I reset main_frame_pos in the slic3r.ini from
_main_frame_pos = 1912,-8_ 
back to 
_main_frame_pos = -8,-8_
it starts up just fine.
I already tried updating my graphics card driver but that didn't help.

bubnikv, 2017-12-05:
This is weird. I tried the same scenario (at least to the extent I understand your description) and everything works correctly.

Do you have monitors of the same size or with different resolution? I have a laptop and an external display, they have different resolution.

Would you please try to start the slic3r-console.exe from a command line to check whether the Slic3r will emit any error message?

Thanks.

jnbr18, 2017-12-09:
I can confrm this bug still exists in version 1.38.3-prusa3d-win64.

I've got two different size monitors but with the same resolution.
I also could not reproduce this on a different computer (different monitor size and resolution).

slic3r-console.exe doesn't output anything, but I have added a Screenshot of how far it gets when its starts maybe that will help you.
[image]

**And just to be sure here's a step by step description**
Step1: Start Slic3r.
Step2: Move the Slic3r window to the 2nd monitor (Slic3r still works fine at that point).
Step3: Close the Slic3r window on the 2nd monitor.
Step4: If you start Slic3r again, Windows will say it's not responding (see Screenshot).

FidelCapo, 2018-12-12:
This should be fixed in Slic3r PE 1.42 alpha, which is to be released soon. Closing.

closed 2018-12-12 by FidelCapo

FidelCapo, 2018-12-12:
Please retest it with Slic3r PE 1.42 alpha when it will be released. If you will have this issue again feel free to reopen. Thank you.