prusa3d/PrusaSlicer #538: Add support for executing end G-code on abort

petterreinholdtsen, 2017-10-07:
Passed on from https://bugs.debian.org/870210 for slic3r-prusa version 1.36.0+dfsg-1:

It would be nice if it were possible to execute the custom End G-code
when a print is aborted.  I end up doing all the steps manually
anyway, and it would add a bit of convenience.

bubnikv, 2017-10-16:
This is not really something for Slic3r, but for the firmware. Slic3r could attach a piece of G-code with the abort section to the resulting G-code file, but the firmware would need to understand that.

As far as I am known, there is no such 3D printer around understanding an abort subprogram.

bubnikv, 2017-10-17:
We may implement the M98 & M99 codes, but that is really a firmware issue
https://www.cnccookbook.com/m98-m99-g-code-cnc-subprograms/

I am closing this issue. For the Prusa Research printers, this feature may be implemented into the firmware.
https://github.com/prusa3d/prusa-firmware

petterreinholdtsen, 2017-10-17:
I'm sorry, but I do not really understand enough about gcode and slic3r-prusa to understand what you are saying.  What can we pass on to the Debian user submitting the original bug.

bubnikv, 2017-10-17:
> What can we pass on to the Debian user submitting the original bug.

This discussion :-)

The submitter may discuss his issue with me here.

His idea is sound, but it cannot be implemented in Slic3r without a support in his printer firmware. Once it is in the firmware, one may or may not implement some support into Slic3r. In my opinion, no support will be required for Slic3r, but it may be convenient.

hyperair, 2017-10-17:
Slic3r Prusa has some G-code streaming functionality these days, doesn't it? Shouldn't that part be able to execute G-code upon cancellation?

bubnikv, 2017-10-17:
> Slic3r Prusa has some G-code streaming functionality these days, doesn't it? Shouldn't that part be able to send the end G-code upon job cancellation?

You mean running over octoprint, or printing online from Slic3r over serial line?

Now I see I may have missed the original intent. We at Prusa Research are printing from SD cards only apart from brief tests of OctoPrint integration.

petterreinholdtsen, 2017-10-17:
[bubnikv]
> The submitter may discuss his issue with me here.

OK.  I sent an update to the Debian BTS asking him to join this
discussion.

-- 
Happy hacking
Petter Reinholdtsen

JoePfeiffer, 2017-10-17:
Hi, I'm the guy who made the original request -- thank you for the discusssion, and thanks Petter for inviting me to join.

I'm running slic3r-prusa3d, running in the GUI and communicating over the serial interface (I find that a *lot* more convenient than messing with the SD card).  My end g-code has some cleanup operations -- retract some filament, raise the extruder, move the bed forward, shut off the bed and hotend.

It would be more better for me if, when I abort a print in the GUI interface, this g-code were sent to my printer so an "abort" did a more orderly shutdown.

I can see the argument that abort should mean "right now", but of course any g-code already sent and buffered is executed on an abort; this can be several seconds of operation.  When I want to shut things off in an emergency, I hit the power switch on the printer!

Anyway, thanks for considering this.

bubnikv, 2017-10-17:
We were actually discussing a similar request at Prusa Research internally
for our Multi Material upgrade kit, where 4 filaments are fed into a 4-way
Y splitter, one at a time. At the abort time, we want the active filament
to be retracted and parked, to be ready for the next print. So your request
is relevant. Only we planned to do it in firmware only.

We don't recommend our customers to use the USB printing for reliability
reasons: If your computer stutters or falls asleep, the print will likely
be ruined or at least there will be some visible scars where the print has
been paused.

We currently don't have enough resources to polish the USB printing as of
now, but we plan to extend our Slic3r team, so your request may get
implemented one day.

JoePfeiffer, 2017-10-17:
I haven't encountered any issues with the USB interface (FWIW, my desktop machine is a Linux server).

lukasmatena, 2024-02-18:
Closing as outdated. The feature to control the printer over USB is no longer present in PrusaSlicer.

closed 2024-02-18 by lukasmatena