prusa3d/PrusaSlicer #540: Separate XY separation settings for support and interfaces

jindrichbenes, 2017-10-09:
XY separation settings for interface layers and main supports will need to be separate, for best results when printing with soluble interface layers.

MStylesMS, 2017-10-10:
Thank your for the attention to this.  I concur that separate settings are needed.  Settings of 30% for soluble supports and 120% for non-soluble supports do appear to work for me as well, in terms of default values.

srepmub, 2017-12-15:
I have trouble understanding how to get optimal results without separate settings at all..? setting it at 0 gives full support, while welding the non-interface support to my parts, but setting it higher will (only in my mind?) degrade the support because of introducing holes in the interface..

jhnphm, 2024-12-16:
Using PETG/soluble supports for PLA for interfaces on Prusa XL gets defeated due to the XY separation as seen below:

[image]

Better illustrated here (but doesn't need the support)
[image]

Workaround is to just use PETG for everything, not just interfaces, since it's cheap enough and set X/Y sep to 0 but it's still not ideal due to the additional toolchanges and priming.

jhnphm, 2026-09-08:
Still an issue on 3.0

@ xy-sep = 80%
[image]
Settings:
[image]

If I change xy-sep to 0, that fixes the gap 
[image]
but causes issues if non-soluble filament was used here:

[image]