prusa3d/PrusaSlicer #542: Feature Request

theopensourcerer, 2017-10-11:
### Version
1.31.7

### Operating system type + version
Ubuntu 16.10

### Behavior
Every time I make a change to a setting to see if I can improve the quality of my prints I always, think to myself "hmm, I should really make a note of these changes". But of course, I rarely do.

_Is this a new feature request?_
Yes.

It would be awesome (to me anyway) if the ini files could be stored in a local (or on Github) repo. So that each time you make a concious decision to make a change you can, choose another option: as well as simply to save the profile, you can also choose to commit the change with a commit message to your repo. Then as a later feature would be an integrated UI so that you could choose to roll-back if you wanted to.

Just an idea that has been bugging me for a while now.

Thanks for a great tool.

theopensourcerer, 2017-11-13:
Maybe this is only my little pipe dream then ;-)

bubnikv, 2017-11-13:
I don't think this will make it into Slic3r. You hackers could use any external git tool, and Jindra, our Slic3r profile guy, is doing exactly that.

I do envision something a bit different. We at Prusa Research would create a hierarchy of read only profiles, where you, the end  user, would override some of them. Then the Slic3r UI could highlight the changes.

Thinking about it, the Slic3r UI could in theory highlight the unsaved fields, maybe with the original value.

theopensourcerer, 2017-11-13:
Thanks for the follow up.

I think anything that could help users have some kind of history or changelog would be really helpful. But I do understand my request is probably a bit too "geeky" so please feel free to close this if you wish.

bubnikv, 2018-08-02:
We have implemented automatic configuration updates, configuration snapshots. We have now another similar request https://github.com/prusa3d/Slic3r/issues/1110 marked with "volunteer needed" :-)
Let's the newer open, let's close this one.

closed 2018-08-02 by bubnikv