prusa3d/PrusaSlicer #543: printing tilted plan need to have a smaller layer.

xynium, 2017-10-12:
### Version
1.33.8 prusa ed

### Operating system type + version
ubuntu 16.04

### Behavior
Printing tilted plan need to have a smaller layer. According to pythagoras law when next layer come she has a longer distance, the more that the angle is hight.
Problem is that this layer does not melt that well and lower the strength, for example in propeler

bubnikv, 2017-10-21:
Frankly I don't understand. Would you please give more details and try to be more specific? A picture tells more than a thousand words. Thanks.

xynium, 2017-10-22:
Hi,
see the sketch
[image]

bubnikv, 2017-10-22:
I see now. The way Slic3r calculates the print paths and the amount of material extruded is supposed to fill the gaps fully. I have highlighted the gaps with blue triangles on the following image.

[image]

If you have gaps unfilled between the threads, then you either have the extruder calibrated incorrectly (you extrude less material than intended), or your material has a high viscosity, so it does not fill the gaps.

Actually most plastic types have a fairly high viscosity, therefore you will not be able to fully close the gaps, no matter how much pressure your extruder will exert. Funny thing is, the extruder pressure will likely be high enough to lift the print head slightly before it fully fills the gaps.

xynium, 2017-10-22:
Thanks for your response, quite new in 3D printing, so i will try with more plastic through the extruder.
I am actually trying to print a propeller
JP

bubnikv, 2017-10-23:
You may also try to slow down to give the filament more time to be squeezed into the gaps.

I don't think this is a software issue, closing.

Feel free to reopen after you gain some experience with the 3D printing technology. You may invent something and help us push the technology forward :-)

closed 2017-10-23 by bubnikv