prusa3d/PrusaSlicer #546: [REQUEST] Layer by Layer Preview Info Overlays

jrezin, 2017-10-15:
### Version
1.37.1-prusa3d

### Operating system type + version
Windows 10 #1703

### Behavior
[FEATURE REQUEST][INTERFACE]
Overlay Icons in the "Preview" pane showing the current layers' fan speed, extruder and bed temps... and maybe current extrusion speed (good to know for small features optimization)

maybe current used extruder for multiple extruder machines? probably something along the line...

CURRENT LAYER:
------------ |
E0 🌡 200ºC
HB 🌡 90ºC
E0F🔁 20%
E0 ⏬ 20mm/s

This feature would be very useful to analyze the generated g-code, see when the fan is turning on, off and its strength (specially when using auto-cooling); changes in the extruder and bed temperature (to confirm it's working visually when desired).

### TL;DR
display more info about the generated g-code visually, layer by layer, to help optimize and debug prints.

mikko70, 2017-11-15:
Gcode/line number, showing current layer is good option in layer view.
Maybe in pop-up window, if no room for it. Or new tab where current layer gcode is highlighted.

bubnikv, 2018-12-11:
@YuSanka This may be a nice feature.

JohnnyDeer, 2019-10-11:
Hi, thank you for idea. We decided to not implemented it in PrusaSlicer for now.  
I will close this issue in framework of improving the clarity of the issues list. 
Thank you for understanding.

closed 2019-10-11 by JohnnyDeer