prusa3d/PrusaSlicer #547: Cannot run post processing batch file under Win7x64

dharris2112, 2017-10-17:
### Version
1.37.1-prusa3d-win64

### Operating system type + version
Win7 x64

### Behavior
I have a command prompt batch file I use to run gpx (gpxbatch.bat). The contents of the batch file is below:
C:\gpx-2.5.2-win64\gpx -glpv -m fcp %1

When I go to create the gcode file, the file is generated but the x3g is not. There is no error reported.
If I look inside the generated gcode file, the post processing line looks like this:
; post_process = C:\\gpx-2.5.2-win64\\gpxbatch.bat

If I perform the same action on the mainline slic3r release 1.29, the x3g is generated as expected. 
If I look inside the generated gcode file, the post processing line looks like this:
; post_process = C:\gpx-2.5.2-win64\gpxbatch.bat

I have tried removing the path to the batch file but then I get an error about not having permissions to execute the script. I've added the directory path to the batch file to the PATH variable in Windows, which is the same directory as gpx, but that does not make a difference.

I'm guessing the escape sequence ('\') is causing a path problem but I'm not sure how to fix this.

#### STL/Config (.ZIP) where problem occurs
[gpx_postprocessfail_config_slic3rpusa3d1371.zip](https://github.com/prusa3d/Slic3r/files/1391981/gpx_postprocessfail_config_slic3rpusa3d1371.zip)

dharris2112, 2017-10-17:
Sorry, the editor removed removed the double '\' from the first post processing statement. You can see the actual line in the configs I've attached (I attached configs from both Pusa3d branch and mainline).

Thanks,

Dave H.

dharris2112, 2017-10-17:
Arghh! Not having a good day :-(

Double "\\" in the path is what I'm trying to get across.

bubnikv, 2017-10-21:
This is weird. I have tested the script execution on Windows for https://github.com/prusa3d/Slic3r/issues/514 and it worked.

Does the following discussion help? https://github.com/prusa3d/Slic3r/issues/514

How did you enter the post processing script path? By editing the config, or by entering into the UI? It may be, that the mainline slic3r diverged in the way the script paths are stored into the config.

bubnikv, 2017-10-21:
Would you please provide your config.ini?

dharris2112, 2017-10-22:
I did review #514 however I'm not running a perl script but a simple command prompt batch file. I did try the "method" from bug #514 which was to include the path to cmd.exe before the gpx command such as:
C:\Windows\System32\cmd.exe C:\gpx-2.5.2\gpx -glpv -m fcp %1
but this did not solve the problem.

I entered the the batch file name into UI.  See attached screen captures for both 1.29 and 1.37.1.

[image]
[image]

There should be a gpx_postprocessfail_config_slic3rpusa3d1371.zip file attached under STL/Config in the bug. I've included both the 1.29 and 1.37.1 ini files within that zipfile.

Let me know what else you need and thanks for looking into this,

Dave H.

bubnikv, 2017-11-30:
I just tried with the latest alpha release on Windows 10 64bit and it works.
[image]
Would you please try with https://github.com/prusa3d/Slic3r/releases/tag/version_1.38.2 ?

dharris2112, 2017-12-06:
Hi,

Thanks again for working on this.

I tried it on 1.38.2 and 1.38.3. Indeed, it does seem to work when you execute gpx.exe only (as you did above). The problem is I have some gpx parameters to that I run in a batch file (gpxbatch.bat). If I use c:\gpx-2.5.2\gpxbatch.bat, I don't get an error like in 1.37.1 BUT I also don't get an .x3g file.

I went back and modified my batch file to add cmd.exe like from [#514](https://github.com/prusa3d/Slic3r/issues/514) )and from earlier in this thread) but that didn't help. It actually broke execution in 1.29.

Cheers,

Dave H.

dharris2112, 2018-01-02:
Hi again. Just checking in to see if anything new has been found?

Thanks,

Dave H.

bubnikv, 2018-02-12:
I wonder whether the Slic3r PE 1.39.1-alpha will work for you now. From the release log:

Fixed environment variable settings for G-code post processing scritps on Windows, thanks @fredizzimo #674.

dharris2112, 2018-02-15:
Hi,

I tried Slic3r PE 1.39.1-beta. Still the same overall issue. At least with this release, I didn't get to modifying the batch file like in #514 yet but I'll try that next and reply back.

Thanks,

Dave H.

dharris2112, 2018-05-03:
Sorry for the long delay on feedback.

I tried the batch file variation based on #514 but that had no effect. I tried this on the latest version: Slic3rPE-1.39.2-beta.

Thanks

bubnikv, 2018-08-02:
I hope we have finally fixed it, see https://github.com/prusa3d/Slic3r/issues/975
Closing.

closed 2018-08-02 by bubnikv