prusa3d/PrusaSlicer #548: [request] filament remaining tracker

i-make-robots, 2017-10-21:
### Behavior
* not always sure if enough filament left on the roll to do the job.
* slic3r knows filament properties and amount used per print.
* knowing the mass of the roll when it was started, it should be possible to some bookkeeping and tally the qty used, then either 
  * warn filament isn't enough or 
  * offer suggestions to make print fit inside remaining amount or
  * offer prints from thingiverse that fit in the remaining volume (use every part of the buffalo)

related to #130?

_this is a new feature request_

ManuGithubSteam, 2017-10-27:
Very good idea!

But this would only work if:

1. You start counter with a new spool, you can not start with half spool etc
2. There needs to be a spare amount for ozzing, loading filement command and so on.

But in general i really like the idea. Should be easy to implement too :-)

mikko70, 2017-11-15:
This option never work. Better use notebook and add there.

i-make-robots, 2017-11-15:
@mikko70 well, not with THAT attitude.. :)

closed 2026-09-29 by prusaslicer-bot[bot]